nisargjhaveri/WirelessAndroidAutoDongle #1: Can't compile

acuartango, 2023-09-25:
Hello !

Thank in advantage for your job, i would like to test it but i can't compile...

I have a 2w raspberry  and when i compile i get:

..................... LAST LINES : 

rsync -a --ignore-times --exclude .svn --exclude .git --exclude .hg --exclude .bzr --exclude CVS --chmod=u=rwX,go=rX --exclude .empty --exclude '*~' package/skeleton-init-sysv//skeleton/ /home/vagrant/AAWirelessDongle/buildroot/output/rpi2w/target/
>>> skeleton  Extracting
>>> skeleton  Patching
>>> skeleton  Configuring
>>> skeleton  Building
>>> skeleton  Installing to target
>>> linux-headers custom Extracting
gzip -d -c /home/vagrant/AAWirelessDongle/buildroot/dl/linux/linux-0b54dbda3cca2beb51e236a25738784e90853b64.tar.gz | tar --strip-components=1 -C /home/vagrant/AAWirelessDongle/buildroot/output/rpi2w/build/linux-headers-custom   -xf -

gzip: /home/vagrant/AAWirelessDongle/buildroot/dl/linux/linux-0b54dbda3cca2beb51e236a25738784e90853b64.tar.gz: not in gzip format
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[1]: *** [package/pkg-generic.mk:217: /home/vagrant/AAWirelessDongle/buildroot/output/rpi2w/build/linux-headers-custom/.stamp_extracted] Error 2
make: *** [Makefile:23: _all] Error 2

Thank you very much !

nisargjhaveri, 2023-09-26:
This looks like a problem downloading and extracting the raspberry pi kernel itself. Can you see if the mentioned path is as expected and the download happened correctly? Not really sure why this would happen.

Also, if you're looking to simply try this out, you can download a build directly from [releases](https://github.com/nisargjhaveri/AAWirelessDongle/releases) as well.

gamelaster, 2023-10-26:
Can be closed I guess?

nisargjhaveri, 2023-10-26:
Yes, closing this as many of us are able to build and there is no update on this issue for some time.

closed 2023-10-26 by nisargjhaveri