nisargjhaveri/WirelessAndroidAutoDongle #5: Bluetooth is not available

danihidayatx, 2023-10-20:
hello,

i build it using my raspberry pi 4, the wifi is available but the bluetooth is not even discovered. so im reflashing using your release instead of building. but got the same result instead
[image]
[image]

nisargjhaveri, 2023-10-22:
Bluetooth is only available when Android Auto is attempted from the head unit. Also, it is discoverable for around 3 minutes by default. You can have a look if your headunit is starting the Android Auto connection via usb or not. If not, we may have to find another approach on how to make it work.

danihidayatx, 2023-10-22:
yes, i look into autostart script. im testing on hyundai ioniq 5 and it's
not working although it's working on mercedes head unit.

for ioniq im gonna debug it further

thanks

danihidayatx, 2023-10-22:
need further test

closed 2023-10-22 by danihidayatx