nisargjhaveri/WirelessAndroidAutoDongle #7: Dosen't work with ATOTO headunit

nu00, 2023-10-22:
Hi,
I have an ATOTO YS102SL (now out of production) with a "Linux" proprietary OS. I tried your project on an RPi4, but it didn't work at all. The head unit didn't even recognize the device. So, I built this fork https://github.com/BluemediaGER/AAWirelessDongle, and it recognizes the RPi, but it doesn't start Android Auto. On the phone (Pixel 8), I get a notification saying "Looking for Android Auto..." and that's all. I also tried the fork on a Peugeot car, and after some tries, it worked fine.
Here are some photos I took of the head unit and the log on a monitor connected to the Pi. Note that every time I tap "Reconnect" on the head unit, I get a new address.
[image]
[image]

BluemediaGER, 2023-10-23:
My fork contains special customizations for my own headunit, which may break functionality for others. Try the main branch of this repo, now that #3 has been merged. If there are still problems that your headunit doesn't recognize the Pi, you can try [this branch of my fork](https://github.com/BluemediaGER/AAWirelessDongle/tree/feature/device-ids).

nu00, 2023-10-24:
I tried both the main branch of this repository and the feature/device-ids branch of your repository, but the head unit still doesn't recognize it. The main branch doesn't even activate Bluetooth, while the feature/device-ids branch activates Bluetooth, labeling it as "OnePlus Nord," but it doesn't accept connections. Unfortunately, I didn't have the display with me today, so I don't know what is actually happening. Is it possible to connect via SSH to the RPi for debugging? I tried, but it asks for a username and password...

danihidayatx, 2023-10-24:
> I tried both the main branch of this repository and the feature/device-ids branch of your repository, but the head unit still doesn't recognize it. The main branch doesn't even activate Bluetooth, while the feature/device-ids branch activates Bluetooth, labeling it as "OnePlus Nord," but it doesn't accept connections. Unfortunately, I didn't have the display with me today, so I don't know what is actually happening on the pi. Is it possible to connect via SSH to the RPi for debugging? I tried, but it asks for a username and password...

yes you can, connect it through the wifi that raspberry pi created, then for the authentication is
username = "root"
password = "password"

nu00, 2023-10-24:
Here's the logs of an attempt to use my ATOTO headunit:

`# cat /var/log/messages
Jan  1 00:00:01 buildroot syslog.info syslogd started: BusyBox v1.36.0
Jan  1 00:00:01 buildroot kern.notice kernel: klogd started: BusyBox v1.36.0 (2023-10-23 20:51:38 CEST)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] Linux version 5.10.92-v7l (user@vmi970725.contaboserver.net) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot 2023.02) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP Mon Oct 23 21:07:06 CEST 2023
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] random: fast init done
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000039bfffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] On node 0 totalpages: 1006592
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 1728 pages used for memmap
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 196608 pages, LIFO batch:63
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   HighMem zone: 809984 pages, LIFO batch:63
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s31628 r8192 d21620 u61440
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s31628 r8192 d21620 u61440 alloc=15*4096
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1004864
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:46:60:BE vc
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] software IO TLB: mapped [mem 0x0000000024890000-0x0000000028890000] (64MB)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Memory: 3840468K/4026368K available (10240K kernel code, 731K rwdata, 2476K rodata, 2048K init, 567K bss, 120364K reserved, 65536K cma-reserved, 3239936K highmem)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]      Tracing variant of Tasks RCU enabled.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x32c/0x500 with crng_init=1
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000035] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[... cut ...]
Jan  1 00:00:03 buildroot user.info usb_gadget: Enabled default usb gadget
Jan  1 00:00:03 buildroot daemon.info : starting pid 284, tty '/dev/tty1': '/sbin/getty -L  tty1 0 vt100 '
Jan  1 00:00:06 buildroot daemon.info rngd: [jitter]: Unable to obtain AES key, disabling JITTER source
Jan  1 00:00:06 buildroot daemon.err rngd: [jitter]: Initialization Failed
Jan  1 00:00:30 buildroot kern.info kernel: [   30.551848] cam-dummy-reg: disabling
Jan  1 00:00:30 buildroot kern.info kernel: [   30.551884] cam1-reg: disabling`

BluemediaGER, 2023-10-25:
> The main branch currently only activates bluetooth after the headunit initiates Android Auto connection, that is the headunit sends the accessory start command, and from your description that might not be happening.

The log looks exactly like this is what happens. The head unit doesn't even try to start Android Auto (by sending a accessory start request).

> The main branch doesn't even activate Bluetooth, while the feature/device-ids branch activates Bluetooth, labeling it as "OnePlus Nord," but it doesn't accept connections.

@nisargjhaveri Looks like this head unit also needs "real" vendor and device IDs.

nisargjhaveri, 2023-10-25:
> Looks like this head unit also needs "real" vendor and device IDs.

Right, looks like it. Any idea if there are vendor and device IDs that we can use freely? Any idea on what some of the legitimate dongles are using?

nu00, 2023-10-25:
> Right, looks like it. Any idea if there are vendor and device IDs that we can use freely? Any idea on what some of the legitimate dongles are using?

looking here (https://source.android.com/docs/core/interaction/accessories/aoa2) the only Product ID for Accessory Mode is 0x2D00. The device ID could be the Google one (0x18d1) since my phone (pixel 8) have the same and it works. Maybe is for the MTP part to be changed (for example use the Nexus/Pixel Device (MTP) 0x4ee1)?
Should the headunit have the ability to look to files on the device in MTP? If so I don't think it's actually showing some files the rpi...

nu00, 2023-10-25:
I don't know if it's related, but connecting the rpi to my PC, windows give me an error (USB\DEVICE_DESCRIPTOR_FAILURE), and looking for the VID and PID it shows USB\VID_0000&PID_0002\5&86C20F0&0&2. Maybe this is the problem? Does the main fork start MTP at first?

BluemediaGER, 2023-10-25:
> Right, looks like it. Any idea if there are vendor and device IDs that we can use freely? Any idea on what some of the legitimate dongles are using?

Depends on what you define as "legitimate", I guess :D
I tested two cheap ones from Amazon, and they both showed up as a Xiaomi phone on the head unit. For more expensive models like the Motorola MA1, my guess would be that they simply use their own IDs.

> I don't know if it's related, but connecting the rpi to my PC, windows give me an error (USB\DEVICE_DESCRIPTOR_FAILURE), and looking for the VID and PID it shows USB\VID_0000&PID_0002\5&86C20F0&0&2. Maybe this is the problem? Does the main fork start MTP at first?

If this happens with the main branch of this repo, I would kind of expected that. It's not a real device, so Windows probably doesn't know how to handle it. My branch with the OnePlus device IDs shoud get detected by Windows just fine, but it will report in device manager that there are no drivers for it. It's enough to trick the head units into thinking that it is a MTP device.

nu00, 2023-10-25:
Yes, I've tested your device-ids branch, and it displays the correct VID and PID in Windows. What I was thinking is that before entering gadget mode, it needs to be recognized as an MTP device with an MTP PID. Correct me if I'm wrong...

nisargjhaveri, 2023-10-26:
Yeah, it does make sense that we need to use some well-known MTP vid/pid, I was misunderstanding something till now. We can use 18d1/4ee1, should work.

I've pushed a change in [device_id](https://github.com/nisargjhaveri/AAWirelessDongle/tree/device_id) branch. Have a look in case I missed anything. We can merge this in main soon, just that I don't have a way to test if these changes will work for these head units.

It is essentially the same change as https://github.com/BluemediaGER/AAWirelessDongle/commit/213d305d279ed38e79d3289ba93d7e1ec5c5052f except I've not changed the manufacturer and product name, I guess that should not be relevant. Also, I've changed the class/subclass/protocol for MTP interface to match better.

nu00, 2023-10-26:
I've tried the your device_id brach and it's still not working on my ATOTO, but we have some progress:
```
# cat /var/log/messages
Jan  1 00:00:01 buildroot syslog.info syslogd started: BusyBox v1.36.0
Jan  1 00:00:01 buildroot kern.notice kernel: klogd started: BusyBox v1.36.0 (2023-10-26 11:52:43 CEST)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] Linux version 5.10.92-v7l (user@vmi970725.contaboserver.net) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot 2023.02) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP Thu Oct 26 12:09:50 CEST 2023
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] random: fast init done
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000039bfffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] On node 0 totalpages: 1006592
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 1728 pages used for memmap
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 196608 pages, LIFO batch:63
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   HighMem zone: 809984 pages, LIFO batch:63
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s31628 r8192 d21620 u61440
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s31628 r8192 d21620 u61440 alloc=15*4096
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1004864
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:46:60:BE vc
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] software IO TLB: mapped [mem 0x0000000024890000-0x0000000028890000] (64MB)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Memory: 3840468K/4026368K available (10240K kernel code, 731K rwdata, 2476K rodata, 2048K init, 567K
bss, 120364K reserved, 65536K cma-reserved, 3239936K highmem)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]      Tracing variant of Tasks RCU enabled.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x32c/0x500 with crng_init=1
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[... cut ...]
Jan  1 00:00:04 buildroot kern.info kernel: [    4.098186] android_work: sent uevent USB_STATE=CONFIGURED
Jan  1 00:00:06 buildroot daemon.info rngd: [jitter]: Unable to obtain AES key,
disabling JITTER source
Jan  1 00:00:06 buildroot daemon.err rngd: [jitter]: Initialization Failed
Jan  1 00:00:30 buildroot kern.info kernel: [   30.551821] cam-dummy-reg: disabling

```

It says USB_STATE=CONNECTED and USB_STATE=CONFIGURED, something has changed.....
Still, I  don't get anything on the headunit...

nisargjhaveri, 2023-10-26:
> while the feature/device-ids branch activates Bluetooth, labeling it as "OnePlus Nord," but it doesn't accept connections.

This is interesting. Can you please share the logs with feature/device-ids branch from @BluemediaGER's fork as well?

nu00, 2023-10-26:
Tomorrow I will do it, since I've to build it again. It's possible that, since I've done that test at work in the parking lot during lunch break, that was someone's phone passing by and coincidentally it only showed up when I had the rpi plugged in...

nu00, 2023-10-27:
Ok, it was just a coincidence. Just tried and it dosen't bring Bluetooth up.

Here the logs:

```

# cat /var/log/messages
Jan  1 00:00:01 buildroot syslog.info syslogd started: BusyBox v1.36.0
Jan  1 00:00:01 buildroot kern.notice kernel: klogd started: BusyBox v1.36.0 (2023-10-23 23:50:28 CEST)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] Linux version 5.10.92-v7l (user@vmi970725.contaboserver.net) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot 2023.02) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP Tue Oct 24 00:07:51 CEST 2023
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] random: fast init done
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000039bfffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] On node 0 totalpages: 1006592
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 1728 pages used for memmap
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   DMA zone: 196608 pages, LIFO batch:63
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000]   HighMem zone: 809984 pages, LIFO batch:63
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s31628 r8192 d21620 u61440
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s31628 r8192 d21620 u61440 alloc=15*4096
Jan  1 00:00:01 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1004864
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:46:60:BE vc
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] software IO TLB: mapped [mem 0x0000000024890000-0x0000000028890000] (64MB)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] Memory: 3840468K/4026368K available (10240K kernel code, 731K rwdata, 2476K rodata, 2048K init, 567K bss, 120364K reserved, 65536K cma-reserved, 3239936K highmem)
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000]      Tracing variant of Tasks RCU enabled.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Jan  1 00:00:01 buildroot kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x32c/0x500 with crng_init=1
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:01 buildroot kern.info kernel: [    0.000036] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[... cut ...]
Jan  1 00:00:03 buildroot kern.info kernel: [    3.451896] dwc2 fe980000.usb: bound driver configfs-gadget
Jan  1 00:00:03 buildroot daemon.info : starting pid 288, tty '/dev/tty1': '/sbin/getty -L  tty1 0 vt100 '
Jan  1 00:00:06 buildroot daemon.info rngd: [jitter]: Unable to obtain AES key, disabling JITTER source
Jan  1 00:00:06 buildroot daemon.err rngd: [jitter]: Initialization Failed
Jan  1 00:00:30 buildroot kern.info kernel: [   30.551798] cam-dummy-reg: disabling
Jan  1 00:00:30 buildroot kern.info kernel: [   30.551835] cam1-reg: disabling

```

nu00, 2023-10-27:
I was thinking: is there any way to sniff the USB traffic between my phone and the head unit? This way, we can better understand what's happening. Do you have any ideas on how to do it?

gamelaster, 2023-10-27:
@nu00 for that, you need to have hardware level USB packet sniffer, I don't know about any cheap one. 150$ for [Cynthion](https://www.crowdsupply.com/great-scott-gadgets/cynthion). But I don't think you need this. The communication is known.

BTW @nu00 Do you still have that Windows USB descriptor issue? Because that would then give a sense why it is problematic.

nu00, 2023-10-27:
No,with @BluemediaGER branch features/device-id in windows it shows up as USB\VID_22D9&PID_2764&REV_0510, but with the device-ids branche of the main build by @nisargjhaveri it shows up as USB\VID_1D6B&PID_0105\0 that are not the 18d1 and 4ee1 @nisargjhaveri set in aa_wireless_dongle/board/common/rootfs_overlay/etc/init.d/S92usb_gadget

nisargjhaveri, 2023-10-27:
> I was thinking: is there any way to sniff the USB traffic between my phone and the head unit? This way, we can better understand what's happening. Do you have any ideas on how to do it?

You may be able to use https://github.com/AristoChen/usb-proxy on a RPi4. I had used it once for something with some modifications. Thought this can be slightly tricky since for AA as the usb resets in the middle and you need to restart the proxy.

We do know how the communication is supposed to happen, but since it is not working on your head unit, might help to know what we're missing.

> but with the device-ids branche of the main build by @nisargjhaveri it shows up as USB\VID_1D6B&PID_0105\0 that are not the 18d1 and 4ee1 @nisargjhaveri set in S92usb_gadget

Are you sure the build went through correctly, 1D6B/0105 is the one we're using in the main branch. Also, you might not need a full rebuild, just ssh into RPi and you can make the same changes as ca7859b2e49b1d30afe32467775b2f6caa5ed7e3 directly.

nu00, 2023-10-27:
Okay, something went wrong with the build, so I manually applied the changes from commit ca7859b, and now Windows shows USB\VID_1D6B&PID_0105\0123456. I'll let my laptop charge up a bit, and then I'm going to try it.

nu00, 2023-10-27:
> You may be able to use https://github.com/AristoChen/usb-proxy on a RPi4. I had used it once for something with some modifications. Thought this can be slightly tricky since for AA as the usb resets [...]

I'm going to give it a try tomorrow, at least we now what's going up...

nu00, 2023-10-27:
@nisargjhaveri, I tried it with the changes from commit ca7859b, and IT WORKS! I apologize for my incompetence with the building; these kinds of projects are new to me, and I'm trying to learn new things. Also, thank you for your work; I've tested it for a while, and I didn't notice any lags in general. I'm going to keep it plugged into my car and try it in the coming days. I'll let you know if something doesn't work.

nisargjhaveri, 2023-10-28:
That's great news! I've merged the `device_id` branch to `main` now as well.

nu00, 2023-10-29:
Hi, I tested it for a while this weekend, and I found out that sometimes, after a while, it disconnects, and it doesn't accept a reconnection from the head unit because the RPi stays in gadget mode and doesn't go back to MTP. It's sporadic; sometimes it happens after a few minutes, and sometimes after 10/15/20 minutes. The only thing I can do is unplug the RPi from the head unit, reset the head unit with a paperclip (because while I'm driving I can't turn off the car, ahaha), and then plug the RPi in again. Here are the logs when it fails:

```
Jan  1 00:02:01 buildroot user.info aawgd[257]: -1 bytes read from USB
Jan  1 00:02:01 buildroot user.info aawgd[257]: Read from USB failed: Input/output error
Jan  1 00:02:01 buildroot user.info aawgd[257]: -1 bytes written to USB
Jan  1 00:02:01 buildroot user.info aawgd[257]: Write to USB failed: Input/output error
Jan  1 00:02:01 buildroot user.info aawgd[257]: Forwarding stopped
Jan  1 00:02:01 buildroot daemon.err bluetoothd[161]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Jan  1 00:02:01 buildroot kern.info kernel: [  121.711619] android_work: sent uevent USB_STATE=DISCONNECTED
Jan  1 00:02:01 buildroot kern.info kernel: [  121.760862] dwc2 fe980000.usb: new device is full-speed
Jan  1 00:02:01 buildroot daemon.err bluetoothd[161]: src/profile.c:ext_io_disconnected() Unable to get io data for HSP HS: getpeername: Transport endpoint is not connected (107)
Jan  1 00:02:01 buildroot daemon.err bluetoothd[161]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:02:01 buildroot user.info aawgd[257]: Bluetooth adapter was powered off
Jan  1 00:02:01 buildroot user.info aawgd[257]: USB Manager: Disabled all USB gadgets
Jan  1 00:02:01 buildroot kern.info kernel: [  121.983283] android_work: did not send uevent (0 0 00000000)
Jan  1 00:02:03 buildroot user.info aawgd[257]: USB Manager: Enabled default gadget
Jan  1 00:02:03 buildroot kern.info kernel: [  123.983586] dwc2 fe980000.usb: bound driver configfs-gadget
Jan  1 00:02:03 buildroot kern.info kernel: [  123.986910] read descriptors
Jan  1 00:02:03 buildroot kern.info kernel: [  123.986943] read strings
Jan  1 00:02:03 buildroot kern.info kernel: [  123.987013] android_work: did not send uevent (0 0 00000000)
```

gamelaster, 2023-10-29:
@nu00 actually, I had issues with dropping of the connection, and I suspect that sometimes, the exact channel is so much busy, that it drops the connection, but it is only suspicion, I don't have this confirmed. Does this happen in cities?

nu00, 2023-10-29:
@gamelaster, to be honest, it happens regardless of where I am. I also noticed that the rpi wifi network is enabled on both 5 GHz and 2.4 GHz. Maybe the phone tries to switch frequencies, and it crashes. However, if you read the logs, it says that it failed reading from USB, so I don't know...
A nice workaraund will be to go back to MTP, so we can reconnect easier from the HU

nu00, 2023-10-29:
I've just tried also to disconnect Android Auto from the notification on the phone, and the logs are the same, so maybe @gamelaster is right.

nisargjhaveri, 2023-10-29:
We do start the MTP again if it fails. Even in the logs, see these lines, which says that the MTP gadget was started again.
```
Jan  1 00:02:01 buildroot user.info aawgd[257]: Bluetooth adapter was powered off
Jan  1 00:02:01 buildroot user.info aawgd[257]: USB Manager: Disabled all USB gadgets
...
Jan  1 00:02:03 buildroot user.info aawgd[257]: USB Manager: Enabled default gadget
```

Two things to investigate.
1. Why is the usb read failing in the first place.
2. Why is it not reconnecting after we try to reset the gadget to MTP mode.

nu00, 2023-10-29:
I don't know why, but I tried to reboot the RPi via SSH (so no unplug and replug), and after some retries, it started reconnecting to the head unit. Notably, I don't have to touch the reconnect button on the head unit, it makes it worse.
It's also hard to debug this issue because it happens randomly

nu00, 2023-10-29:
what abaout these line?
`bluetoothd[161]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down`

seems like the bluetooth crashes...

nu00, 2023-11-09:
> Two things to investigate.
> 
> 1. Why is the usb read failing in the first place.
> 2. Why is it not reconnecting after we try to reset the gadget to MTP mode.

1. After some testing, I noticed that it's the HU fault for the usb disconnection, it happen also with the phone connected via usb directly. 
2. For the reconnection, I'm not 100% sure, but I think the phone goes in "charging mode" before going to MTP. I would give it a try but I really don't know how to implement it (some advice pls)

nisargjhaveri, 2023-11-11:
> For the reconnection, I'm not 100% sure, but I think the phone goes in "charging mode" before going to MTP. I would give it a try but I really don't know how to implement it (some advice pls)

We do disable the gadget completely for 2 seconds, which should be equivalent to the "charging mode" before going into MTP. We can try to tweak with the time limit if that's the issue. https://github.com/nisargjhaveri/AAWirelessDongle/blob/main/aa_wireless_dongle/package/aawg/src/aawgd.cpp#L33C14-L33C14

From the logs above, it looks like we do enable the MTP again, but the headunit never sends the accessory start request

nisargjhaveri, 2024-04-01:
We have more robust reconnection now with v0.4.0. Resolving this issue for now, feel free to reopen if it still is an issue.

closed 2024-04-01 by nisargjhaveri