nisargjhaveri/WirelessAndroidAutoDongle #8: Port to Java

hkfuertes, 2023-10-24:
Hello, I would love to have a try and replicate the bluetooth/wifi communication to android/java/kotlin to make the AAGateWay work as if it was native, but I don't find how it is done, no architecture, no point to start... how is the protocol? Is it wifi or wifi direct? Bluetooth socket in, response with the ip, the phone connects to the ip?

If you could provide some info or doc or point to start digging, it would be awesome.

Thank you very much!

nisargjhaveri, 2023-10-24:
There are already some android/java/kotlin implementation. You can have a look at [nisargjhaveri/AAWirelessGateway](https://github.com/nisargjhaveri/AAWirelessGateway) for a Kotlin implementation. You can also have a look at [borconi/AAGateWay](https://github.com/borconi/AAGateWay) for a more basic Java implementation.

Unfortunately, I don't have any documentation, you can try to go through `AAWirelessProfile` and `AAWirelessLauncher` in [bluetoothProfiles.cpp](https://github.com/BluemediaGER/AAWirelessDongle/blob/main/aa_wireless_dongle/package/aawg/src/bluetoothProfiles.cpp) to have an idea of the bluetooth communication to initiate the connection. Kotlin implementation for the same can be found at [AABluetoothProfileHandler.kt in nisargjhaveri/AAWirelessGateway](https://github.com/nisargjhaveri/AAWirelessGateway/blob/main/app/src/main/java/com/nisargjhaveri/aagateway/AABluetoothProfileHandler.kt).

Hope this helps! :)

hkfuertes, 2023-10-24:
Hahaha, no need, you are awesome! My point was to do exactly that app... but you already did it. 

OK, for now Im testing this project with a zero2w and a pi3a (which also supports OTG, I'm figuring out the `dtb` situation)... my pi0w are all broken.... 

Then I will play with a FireStick that I have with LineageOS18 to convert it into a dongle... maybe auto enabling hotspot on boot, pairing via adb shell... etc... would it be easy to add command line support on that app to configure hotspot details via adb shell?

nisargjhaveri, 2023-10-24:
> Then I will play with a FireStick that I have with LineageOS18 to convert it into a dongle... maybe auto enabling hotspot on boot, pairing via adb shell... etc... would it be easy to add command line support on that app to configure hotspot details via adb shell?

If you can get the app running, it does take care of enabling hostspot and connecting via bluetooth. Are you talking about the one-time configuration since it might not be easy to get the app on screen with the FireStick?

hkfuertes, 2023-10-24:
Yes, I mean, I got root access via `adb shell` so I can do anything from there at any time, but everything has to be headless...

gamelaster, 2023-10-25:
It works as follows: (could be not perfectly accurate, as I am writing this on phone) 
When you connect to Bluetooth, RPi send via RFCOMM info about it's Android Auto (protobuf serialization), with wifi name, password, IP address of server and port. Phone follow this and connects to specified wifi, and connects to TCP server specified before. Afterwards, dongle acts as Android phone in MTP mode, Headunit checks if it supports AOA (android accessory), if yes, then it will tell you to switch to it, you do as follows (AOA have different usb descriptor), Headunit detect it and sends initial handshake packet. You then forward this packet to phone through TCP server, and the respond from phone back to Headunit, and afterwards it just acts as gateway of packets.

If you have any questions, let me know.

hkfuertes, 2023-10-25:
Thank you for the explanation! I was going crazy, trying to understand how the phone can be connected to a hotspot AND not be sending traffic through it... It seems Android Auto can handle a wifi connection without specifying a route.

I have been playing with AAGateWay project, using hotspot in slave, in master... and I was not fully understanding how the process works... I guess that the hotspot created here does not provide any DHCP, therefore no GW, therefore no route... and this "dhcp" is handle by the RFCOMM connection...

Thank you a lot!

hkfuertes, 2023-10-25:
Let me put it in other words... Why if I connect to the hotspot manually, the phone (Pixel 7) wants to route all traffic through hotspot (with no internet)... but if Android Auto connects, it doesn't?

gamelaster, 2023-10-25:
This is just my speculation, but my previous experience with using [WiFi IoT API -> Request Network API](https://developer.android.com/guide/topics/connectivity/wifi-infrastructure), when you get information about the WiFi through RFCOMM, Android treats the WiFi as it is without connectivity, so not only that Android keeps connection until app which uses this WiFi runs, but also tells the network manager to use 4G or any other connectivity source for internet.

hkfuertes, 2023-10-26:
Thank you for the explanation, that makes much more sense! I guess we can close this issue :)

closed 2023-10-26 by hkfuertes