nisargjhaveri/WirelessAndroidAutoDongle #9: Issue with Bluetooth Pi Zero 2W

hkfuertes, 2023-10-24:
Hello, I finally made my Zero 2W bring up Bluetooth. By default it complains that the patch `bcrm/BCM.hcd` is missing. But its right there in `/lib/firmware/bcrm/BCM43430B0.hcd`. I manage to fix this by ssh into the PI and copying it to the new name: 
```bash
cd /lib/firmware/bcrm/
cp BCM43430B0.hcd BCM.hcd
```
So that`s the fix, but I don't know how to integrate it into `Buildroot`, and therefore, not able to open a PR.

gamelaster, 2023-10-25:
I used it on Z2W yesterday, and everything worked perfectly for me.

nisargjhaveri, 2023-10-25:
I've also been using with RPi Zero 2W without any issue. Are you facing the issue even with the prebuilt images?

gamelaster, 2023-10-25:
@hkfuertes On my KIA Niro 2017 it works flawlessly.

BluemediaGER, 2023-10-25:
@hkfuertes My fork changes nothing in that regard, but feel free to try it. I'm also using a Pi Z2W (and a Pi 4 for testing), and I didn't have these problems.

hkfuertes, 2023-10-26:
Sorry I'm late, but yes you where all right, it has to be my board. Yesterday I built a PI3A+ Image with this repo, and it works flawlessly on my KIA. I will investigate, and maybe get another Zero2W, as mine was never used since I bought it on launch (maybe its outdated 🤷‍♂️ ).

I guess we can close this issue. BTW, I will try to make a PR this evening to add the `defconfig` I used for the PI3A+, mainly the Zero2W's with the proper `dtb` (from Buildroot oficial pi3 defconfig) and as this board has 5GHZ, the hostapd from PI4 Overlay. I need to check if in the `dtoverlay=dwc2,dr_mode=peripheral`, 'dr_mode' is required or not, for now I have it, but maybe is not necesary, and then the `defconfig` can be simpler.

(In case anyone is curious, the 3A+ only has a single USB, with no hub, so the controller can also be put in gadget mode, then with a USBA to USBA cable, everything works, even it takes power from there!)

closed 2023-10-26 by hkfuertes

niravatgit, 2023-10-29:
> I've also been using with RPi Zero 2W without any issue. Are you facing the issue even with the prebuilt images?

Please share link/location for the prebuilt images. I built it yesterday and was testing it today, but Bluetooth is not coming up. #13

hkfuertes, 2023-10-29:
I Will try to build one, but if you can build it yourself, I can tell you what to add.
inside `buildroot/board/raspberrypi/post-build.sh` add at the end:
```shell
cd ${TARGET_DIR}/lib/firmware/brcm
if [ ! -f BCM.hcd ]; then
    ln -s BCM43430B0.hcd BCM.hcd
fi
```

nisargjhaveri, 2024-01-22:
This should now be fixed by 8c0dd4c42ee6d860f82895671bff2dca9de97c03.