nisargjhaveri/WirelessAndroidAutoDongle #10: Orange Pi zero 2W support

jobitjoseph, 2023-10-26:
The new Orange pi Zero 2W looks like the perfect hardware for the use. It provides much better hardware spec with respect to the price. If possible kindly add support for this particular board. Thank you

nisargjhaveri, 2023-10-26:
Yes, it does look promising. Unfortunately, I don't have one yet to test. Happy to take in PRs if someone is willing to test it out. Also, not sure if buildroot already has a defconfig for it yet.

gamelaster, 2023-10-26:
It would be possible, mainline have USB OTG driver, although, it's important to have DMA working on the driver, without that, the performance might be very slow. I asked about it, but I very doubt that it is done.

if DMA is here, I might consider taking look on it, but RPi Zero 2 W is cheap, and performance is sufficient, so what advantage would this one give us?

gamelaster, 2023-10-26:
Addition: confirmed that, there is not mainline DMA driver for USB OTG, that means low performance and high CPU usage, what is big problem for this usage. So not worth to do anymore.

hkfuertes, 2023-10-26:
Maybe this could help? He is trying to bring up mass storage: http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=145208

gamelaster, 2023-10-26:
@hkfuertes the USB OTG driver is there in mainline, and it works, problem is, that there is no Direct Memory Access support, so every write / read needs to be manually done by CPU, instead of specifying buffer by CPU and then instructing DMA to give it to USB OTG IP Core. This means, that CPU will waste a lot of cycles (thus high cpu usage), and for big data, it will be not that fast as when it's done by DMA. From what I know, the max speed of it is like 12Mbps, what is definitely not sufficient. [Some more info here](https://forum.armbian.com/topic/23510-usb-otg-help-needed/?do=findComment&comment=147103)

jobitjoseph, 2023-10-28:
will this be helpful? https://elixir.bootlin.com/linux/v6.6-rc7/source/drivers/usb/musb/sunxi.c

gamelaster, 2023-10-28:
@jobitjoseph this is the driver, but when you notice DMA functions, whose should return the DMA implementation, they returns null, thus, no DMA support:
[image]

closed 2023-11-29 by jobitjoseph