nisargjhaveri/WirelessAndroidAutoDongle #13: Bluetooth device not showing up on pixel 7

niravatgit, 2023-10-29:
I am using rpizero2w, wifi ssid is showing up however Bluetooth device isn't showing.up. any particular config that need to be updated? Any specific Bluetooth config need to be updated post installation? Thanks

gamelaster, 2023-10-29:
Works OK with my Z2W and Pixel 7. Does head unit see the USB device? Only after that the Bluetooth device is visible (on latest release)

niravatgit, 2023-10-29:
Thanks @gamelaster . I see let me check this, I was under impression that BT is enabled as soon as z2w is up.

hkfuertes, 2023-10-29:
I had somewhat a trouble with Bluetooth myself as well... Can you hook a display to the Zero2W? If an error showing `BCM.hcd not found` appears, here is a solution... https://github.com/nisargjhaveri/AAWirelessDongle/issues/9

niravatgit, 2023-10-29:
@hkfuertes thank. Though, I don't have a display to connect to my pi Z2W, I will ssh to it and see if I am able to bring up Bluetooth interface.

NegussugeN, 2023-12-20:
> @hkfuertes thanks. Though, I don't have a display to connect to my pi Z2W, I will ssh to it and see if I am able to bring up Bluetooth interface.

I believe I am having a similar issue. How are you going a out connecting via ssh? I see that regardless of Bluetooth functionality a wifi network is created, but I'm not sure what the password for it is.

nisargjhaveri, 2023-12-27:
The default wifi password is "ConnectAAWirelessDongle" as defined in [hostapd.conf](https://github.com/nisargjhaveri/AAWirelessDongle/blob/main/aa_wireless_dongle/board/common/rootfs_overlay/etc/hostapd.conf). You can ssh using root/password as username/password for now, defined in relevant defconfigs, e.g. [raspberrypi0w_defconfig](https://github.com/nisargjhaveri/AAWirelessDongle/blob/main/aa_wireless_dongle/configs/raspberrypi0w_defconfig).

ProtectiveManEgg, 2024-02-04:
Not sure if this is solved, but same scenario. I've got a Pixel 7 and I didn't see the bluetooth device but I did see the SSID. I tried a different cable to connect the RPI 0w2 and it finally showed up. It ended up being the cable falsely advertised as data capable

closed 2024-04-01 by nisargjhaveri