nisargjhaveri/WirelessAndroidAutoDongle #14: Xiaomi cant connect to bluetooth

leifj79, 2023-11-10:
Have an xiaomi device and i cant connect to the rpi 2 zero by bluetooth using the latest code! 
Have tested BluemediaGER fork and there i can connect and it shows up correctly but then it cant connect to my cars android auto. 
The bluetooth doesn't show up unless i tell my phone to not show names on bluetooth and then i can see the mac adress of the device but pairing fails!

nu00, 2023-11-11:
Try to build the project from source, with #7 have been made some changes in VID and PID of the USB mode that might help you with your problem

leifj79, 2023-11-13:
Im building from ca7859b and will be back with logs

nisargjhaveri, 2023-11-13:
You can also try the [latest build on main from Github Actions](https://github.com/nisargjhaveri/AAWirelessDongle/actions).

nisargjhaveri, 2023-11-17:
The logs suggests that the headunit is not recognising and not sending the accessory start request. I'm out of ideas on why this would happen if Android Auto works in the car, as we're very close to what Pixel devices report via USB.

leifj79, 2023-11-18:
The head unit shows it's connected to Android Auto

leifj79, 2023-11-20:
I guess its my head unit that is to blame, i have to choose between android auto and mirrorlink and i think that is the issue :(
haven't found any way to disable mirrorlink in the car, but newer models dosnt have mirrorlink and it seems fine!

nisargjhaveri, 2023-11-21:
> I guess its my head unit that is to blame, i have to choose between android auto and mirrorlink and i think that is the issue :(

It is technically supposed to work once you select Android Auto. I guess #2 also discusses about a headunit that has similar options for Android Auto and MirrorLink.

leifj79, 2023-11-21:
Yeah the BT starts and my phone tried to connect but doesn't connect before the head unit resets

nisargjhaveri, 2023-11-22:
Right, you can maybe try out the suggestion here: https://github.com/nisargjhaveri/AAWirelessDongle/issues/2#issuecomment-1783434263

Though, this looks like a common issue. Let me try to make a permenant change to fix the timeouts.

leifj79, 2023-11-27:
Tested BluemediaGER version and it works, now I'm going wireless with my Passat 🥳

leifj79, 2023-12-27:
Works with suggested solution

leifj79, 2024-01-21:
Had some new problems but I built from the tree optimistic_phone_connection and my wolksvagen accepts it without any issues! Awsome work!