nisargjhaveri/WirelessAndroidAutoDongle #16: No bluetooth with Raspberry Pi 0 w

aytacortac, 2023-11-13:
Wifi okay but no bluetooth on list.

nisargjhaveri, 2023-11-14:
This is likely not an issue with the bluetooth as such, but due to some reason headunit is not recognising the dongle as phone and not sending the accessory start request.

We start bluetooth only after the dongle receives an accessory start request indicating that the Android Auto is available and initiated from the headunit.

Which build did you try? If you used the build from release v0.1.1, try using one of the [latest builds from main branch](https://github.com/nisargjhaveri/AAWirelessDongle/actions?query=branch%3Amain). There are some additional changes in main that makes it work on more headunits.

If that also doesn't work, try to collect some logs from the RPi using ssh.

closed 2023-11-15 by aytacortac

nisargjhaveri, 2023-11-22:
This looks like a timeout issue, I'll try to find a solution to avoid the timeouts. Thanks for reporting.