nisargjhaveri/WirelessAndroidAutoDongle #19: Compatibility for a arm Mobile?

alexplayzzz, 2023-11-18:
I used to have custom rom on my old device which has good specs is there a way I could flash the file of raspberry Pi & able to use my phone has AA Dongle

nisargjhaveri, 2023-11-18:
You might be interested in this project that works on Android: https://github.com/nisargjhaveri/AAWirelessGateway

I don't think it would be easy to flash the RPi build on a phone, or anything else but RPi very easily. If the board has support in buildroot, it might be possible to try to create a build, but otherwise it would be difficult.

nisargjhaveri, 2023-11-21:
Closing this as there are other projects that works for Android and you're better off using one of those on a phone.

closed 2023-11-21 by nisargjhaveri