nisargjhaveri/WirelessAndroidAutoDongle #20: Difficult Connection

aytacortac, 2023-11-19:
My car is Toyota Corolla 2023 (Europen version, only wired Android Auto). My phone is Samsung Galaxy S20 FE.

I can barely connect to the Raspberry Pi. It rarely connects the first time. I need to try a second or third time. I have to restart the multimedia unit many times. Only then can it establish a connection. It does not disconnect once connected.

What can be done to connect it at once?

nisargjhaveri, 2023-11-22:
From what you explained, very likely the head unit is timing out before we finally connect everything.

You can try this suggestion here: https://github.com/nisargjhaveri/AAWirelessDongle/issues/2#issuecomment-1783434263

If that doesn't work, we can try to do a handshake early without waiting for everything to connect.

nisargjhaveri, 2023-12-05:
You can also check if the builds from #25 helps with more reliable connection in your case?

nisargjhaveri, 2024-01-20:
You can also try out #35, should help with the timeouts.

nisargjhaveri, 2024-03-25:
There have been multiple changes and I believe this issue should be fixed in latest builds. Closing this for now.

closed 2024-03-25 by nisargjhaveri