nisargjhaveri/WirelessAndroidAutoDongle #21: AA bluetooth profile

Demon000, 2023-11-21:
Hi. Having looked at your implementation, I think you'd be better off by not connecting using HSP (since you don't really handle it)  but rather providing a complete ServiceRecord property when registering the AA profile.

This could also leave the door open for using ofono to handle HSP properly I guess, besides cleaning up the code and making it make more sense.

(output of `sudo dbus-monitor --system` attached)

```
method call time=1700571187.703631 sender=:1.406 -> destination=org.bluez serial=4 path=/org/bluez; interface=org.bluez.ProfileManager1; member=RegisterProfile
   object path "/com/aa/aa"
   string "4de17a00-52cb-11e6-bdf4-0800200c9a66"
   array [
      dict entry(
         string "Name"
         variant             string "AA Wireless"
      )
      dict entry(
         string "Role"
         variant             string "server"
      )
      dict entry(
         string "Channel"
         variant             uint16 8
      )
      dict entry(
         string "ServiceRecord"
         variant             string "<?xml version="1.0"?>
<record>
    <attribute id="0x0001">
        <sequence>
            <uuid value="4de17a00-52cb-11e6-bdf4-0800200c9a66"/>
            <uuid value="0x1101"/>
        </sequence>
    </attribute>
    <attribute id="0x0003">
        <uuid value="4de17a00-52cb-11e6-bdf4-0800200c9a66"/>
    </attribute>
    <attribute id="0x0004">
        <sequence>
            <sequence>
                <uuid value="0x0100"/>
            </sequence>
            <sequence>
                <uuid value="0x0003"/>
                <uint8 value="0x08"/>
            </sequence>
        </sequence>
    </attribute>
    <attribute id="0x0005">
        <sequence>
            <uuid value="0x1002"/>
        </sequence>
    </attribute>
    <attribute id="0x0009">
        <sequence>
            <uuid value="0x1101"/>
        </sequence>
    </attribute>
    <attribute id="0x0100">
        <text value="WebAuto Bluetooth Service" encoding="normal"/>
    </attribute>
    <attribute id="0x0101">
        <text value="AndroidAuto WiFi projection automatic setup" encoding="normal"/>
    </attribute>
    <attribute id="0x0102">
        <text value="WebAuto" encoding="normal"/>
    </attribute>
</record>
            "
      )
   ]
```

Demon000, 2023-11-21:
> https://github.com/jcwenger/openauto/blob/c3f9c29ce70357bec9c97c2f4a1b13762919bec1/btservice/AndroidBluetoothService.cpp#L38?

This is how the SDP record is constructed when using QT. For DBus you'd have to provide the ServiceRecord property as a string containing XML. Also, the Channel and Role needs to be provided when using DBus, so the RFCOMM server gets registered. QT uses a legacy method for that.

nisargjhaveri, 2023-11-21:
How does this help? I was under the impression (with some preliminary testing) that Android only reacts to HFP connection and looks for Android Auto service only if a HFP sink is connected. Does the extra attributes defined in this service record make it work without having to connect HFP? That would in fact be very beneficial if it works for all!

Demon000, 2023-11-21:
> How does this help? I was under the impression (with some preliminary testing) that Android only reacts to HFP connection and looks for Android Auto service only if a HFP sink is connected. Does the [...]

That's exactly the case.

I could maybe make a PR that would help you with the changes, but my C++ is rusty.

Demon000, 2023-11-21:
#22

nisargjhaveri, 2024-03-01:
#72 would achieve not having fake HSP profile.

nisargjhaveri, 2024-03-30:
Fixed with #72

closed 2024-03-30 by nisargjhaveri