nisargjhaveri/WirelessAndroidAutoDongle #23: Re-connect doesn't work properly

gamelaster, 2023-12-02:
(will add description later, just adding it on phone so I don't forget :D)

GreenZeeGuy, 2023-12-03:
Many thanks to the global team for this fantastic project!!
I have been facing the reconnect failing issue a few times with the v0. 2.0 release .
Looking forward to the new release with the fixes!

nisargjhaveri, 2023-12-04:
@gamelaster, @GreenZeeGuy Could you please add some more info on the issue you're facing? What do you mean by reconnect in this case?

gamelaster, 2023-12-04:
@nisargjhaveri Of course I forgot, sorry :rofl: . The thing is, that sometimes, when connection drops, or initial connection times out, the re-connect mechanism (if there is any for specific situation), will not kick in, and only solution is to re-plug the device.

GreenZeeGuy, 2023-12-04:
@nisargjhaveri I am facing the same issue that @gamelaster mentioned.

nisargjhaveri, 2023-12-05:
For the initial connection timeout, there is no re-connect mechanism as of now, need to fix this.
If the connection drops due to USB failure, it is supposed to reconnect. TCP failures are not yet handled very well. Though, if you can get the logs out when it fails to reconnect after connection drop, it'd be super useful to know which scenario is failing.

nisargjhaveri, 2024-03-25:
The initial connection failures are handled better with some of the recent commits in main. And handling disconnection after a successful connection is tracked with #26.

Closing this for now. Feel free to reopen if you still face this with builds from latest main.

closed 2024-03-25 by nisargjhaveri