nisargjhaveri/WirelessAndroidAutoDongle #24: Looking for Android auto

alexplayzzz, 2023-12-03:
Since my interest in this project, at first I was wondering to run this in arm but due to board compability & your feedback I planned on ordering a raspberry Pi zero w, I flashed the img into sd & all the process done once I start the device I connect to the Bluetooth sometimes it doesn't connect to the wifi & if connected it still loops on with notification looking for Android auto device which is annoying I tried it with head unit of Suzuki Scross & Head unit reloaded tell me if I am doing any mistakes

alexplayzzz, 2023-12-03:
It doesn't connect to wifi & keeps saying looking for Android auto I am using pixel 6a

alexplayzzz, 2023-12-03:
[image]
[image]

nisargjhaveri, 2023-12-04:
Seems like the wifi connected fine from the screenshot, but the AA is not able to connect. My hunch is that, this happens because the headunit timeouts before the wifi connects.

nisargjhaveri, 2023-12-04:
You can try out the builds from #25 if it helps in this case. I've attempted to work around the timeout by establishing AA communication early with the headunit.

alexplayzzz, 2023-12-05:
I found a better way to fix it I powered the raspberry by the ciggerate port of the car which turns on before turning on headunit & it matches the connection time & in the same interval head unit boots up

closed 2023-12-05 by alexplayzzz