nisargjhaveri/WirelessAndroidAutoDongle #27: Device not compatible in the car

Uneip, 2023-12-12:
I have a MBUX systeme on a mercedes, when i launch android auto on the car, i can't, i have a message witch say "uncompatible device" i have a pixel 7a

nisargjhaveri, 2023-12-19:
Does it work with a wired connection? Also, when do you see the message? Does it start the wireless connection to your phone before showing the message?

Uneip, 2023-12-19:
with my pixel 7a with a usb cable, android auto works. i have installed the pi0w image from your site on a pi zerow 1.1 i have tested with a samsung S20FE to, and so i have the same result "uncompatible device". When the raspberry start, the car lauch android auto, my phone is paired by bluetooth with "AA Wireless Key" and the message box appear with this uncompatible devic text message.