nisargjhaveri/WirelessAndroidAutoDongle #28: Lags in android interface and musics

idarius, 2023-12-12:
Hello, 
i installed the app on a raspberry pi zero W, in a peugeot 2008.
my phone is a Samsung Galaxy S23 ultra, i don't think the problem is coming from there.

The phone connect quickly, no problem to launch the android auto interface.

But there is some lag in the interface, for exemple in waze when i move the map it react 1/2 sec after, and not in a fluid way.
The music (from spotify) got some micro stutter sometimes too, cutting for 1/2 sec and coming back.

Could the Rpi zero W not be beefy enough ?

When connected directly with my phone, i got no problem at all and the interface is totally fluid.

Thx anyway for your app, you saved me from buying a dongle ;)

gamelaster, 2023-12-13:
Hi, it's strange, as it performs pretty well for me, but I have Zero 2W. Anyway, how far you have RPi from your phone? Also, I have some microstutters too, when I am in areas with high density of 2.4 GHz Wi-Fi APs, like cities.

idarius, 2023-12-13:
The Phone is at 20 cm max from the rpi zero.
I will try with a rpi3 that is laying around when I have some time  and report here :) 
Maybe with a rpi 4 and the 5ghz network, it Will maintain a better connection.

gamelaster, 2023-12-13:
@idarius only Raspberry Pi 3A+ is supported. RPi 3 is not. RPi 4 might perform better on 5 GHz

hkfuertes, 2023-12-13:
Yes, Only 3a+ is suported and you would need to connect it through the usb-a port (usb-a to usb-a cable). It has 5ghz (like the 4) and for me its working flawlessly.

idarius, 2023-12-13:
Thanks for thoses informations.
I will try with one or another soon, closing this issue for now !

closed 2023-12-13 by idarius

rskkw, 2024-01-05:
I'm experiencing the same issue with Pi zero w, in a Peugeot5008 with Google pixel 7A.

@idarius Did you have better performance with rpi4? Or did you solve in another way?

idarius, 2024-01-05:
> I'm experiencing the same issue with Pi zero w, in a Peugeot5008 with Google pixel 7A.
> 
> @idarius Did you have better performance with rpi4? Or did you solve in another way?

Hello, I didnt try with another hardware yet.
I dont use it for now because it won't connect with my wireguard connection active, wich I need for my other services.