nisargjhaveri/WirelessAndroidAutoDongle #29: Add the ability to switch devices

AmoghN, 2023-12-19:
Hey Y'all,

I run into this situation where I want to disconnect the current WirelessAA connection and let another device connect instead. Currently, for me to do this I would need to reboot the PI, have my device in airplane mode so that the other device can connect. 

I am wondering if it would be possible to have a feature where switching devices would be easier. Maybe something like if a device is requesting a Bluetooth connection then make a connection to that device instead of the previously connected device 🤔.

Thanks!

nisargjhaveri, 2023-12-19:
Yes, this is something we should have and would make life easier. It would also be possible to do, we might need to store some kind of state for this to work effectively across restarts.

hkfuertes, 2023-12-20:
Is not that controlled here? 
https://github.com/nisargjhaveri/AAWirelessDongle/blob/d6774f5a353174c5e97d5e153650befed0455a20/aa_wireless_dongle/package/aawg/src/bluetoothHandler.cpp#L150-L163

nisargjhaveri, 2023-12-20:
This part falls back to the next device if the first one is not available. But for this to work, as @AmoghN mentioned. The first device's bluetooth needs to be off.

ansher, 2024-01-27:
Hello all, this is great development !

As a suggestion to switch phones:
1. Gpio mounted button, when pressed triggers the script.
2. Script forces bt/wifi disconnect from the existing device.
3. Script requests connection to a next bt device from the list.