nisargjhaveri/WirelessAndroidAutoDongle #30: Ford Fusion Not Working

NegussugeN, 2023-12-21:
After flashing the premade image to a Pi Zero 2W, I am not able to get AA functioning. The car's head unit never responds to anything, nor does a new Bluetooth device appear from the Pi to pair my Pixel 6 Pro with. 

I realize with that information not much help can be provided, so my question: What is the best way to troubleshoot the connection issues? I see that a wifi network is available, so I thought I'd ssh to it to poke around, but I'm not sure what the password is.

nisargjhaveri, 2023-12-27:
The default wifi password is "ConnectAAWirelessDongle" as defined in [hostapd.conf](https://github.com/nisargjhaveri/AAWirelessDongle/blob/main/aa_wireless_dongle/board/common/rootfs_overlay/etc/hostapd.conf). You can ssh using root/password for now, defined in relevant defconfigs, e.g. [raspberrypi0w_defconfig](https://github.com/nisargjhaveri/AAWirelessDongle/blob/main/aa_wireless_dongle/configs/raspberrypi0w_defconfig).

NegussugeN, 2024-01-20:
[AADongle Messages.txt](https://github.com/nisargjhaveri/AAWirelessDongle/files/13996805/AADongle.Messages.txt)

Here is what I'm seeing in messages.  Does anything here look out of the ordinary to you?

nisargjhaveri, 2024-01-20:
This looks like an issue

```
Jan  1 00:00:04 buildroot user.info aawgd[224]: Did not find any bluetooth adapters\
```

I can also see these, might be related.
```
Jan  1 00:00:02 buildroot kern.err kernel: [    1.759758] Bluetooth: hci0: BCM: firmware Patch file not found, tried:\
Jan  1 00:00:02 buildroot kern.err kernel: [    1.762682] Bluetooth: hci0: BCM: 'brcm/BCM.hcd'\
```

Looks like bluetooth is not working properly for your device/build.

nisargjhaveri, 2024-01-20:
See #9, seems to be the same issue.

nisargjhaveri, 2024-01-20:
Did some research, seems like some RPi 2W uses a particular version of the chip that is not handled properly in the kernel version we're using. It seems to have fixed in newer RPi kernels, we might have to try update the kernel or patch our version.

I see that buildroot upstream not also has a new kernel for RPi, so should be feasible to use the new kernel without much effort.

In the meantime, the workaround from #9 might work for you.

nisargjhaveri, 2024-01-22:
This should now be fixed by 8c0dd4c42ee6d860f82895671bff2dca9de97c03.

closed 2024-02-11 by nisargjhaveri

borsosgy, 2024-02-13:
Hi nisargjhaveri,
First of all I would like to say a big thank you for your efforts.

I had the same problem as MrCrrispy.
My LOG was almost the same, I also tested lots of USB cables.
The problem was the same, the head unit timing was bad. After starting the AAWirelessDongle, the Bluetooth connection was not established so quickly and the head unit reported errors.
I made the new build (#35 "optimistic phone connection") on Pi Zero 1.1.(pi0w)
The "optimistic phone connection" function works, but now the Bluetooth communication fails and does not return.
This means that AAWirelessDogle is paired with the cell phone via Bluetooth, --> Android Auto starts automatically on the Headunit, --> the Bluetooth connection between AAWirelessDongle and Phone fails, -->Android Auto started works, but without Bluetooth connection, so the sound doesn't work on the hay unit.

Do you have an idea what I could do?

thanks in advance

Gyula