nisargjhaveri/WirelessAndroidAutoDongle #31: Delay USB switching

fkloft, 2023-12-29:
I'm happy to announce that AAWirelessDongle works with Suzuki Swift (model year 2022) using a Raspberry Pi 4 B. I just ordered a Pi Zero W as a permanent solution.

However, my car often has an issue when the USB is already connected when I turn on the ignition. I already noticed this when using wired AA, so I often wait until the radio has fully powered on (takes a few seconds) before I plug in my phone. It seems the same applies to AAWirelessDongle. When the Pi is already plugged in, the car doesn't connect; if I plug it in after the radio has booted, it works.
I am unfortunately not familiar with the USB protocol and the implementation on the RPi. Do you think it would be possible to artificially delay switching to gadget mode so the car's radio has time to boot?

rcazpeitia, 2024-01-17:
How are you doing?

What kind of radio do you use in the swift?

I have a jimny 2024 and by USB it detects without problems my cell phone, when I connect my Pi zero W, I connect the device to the wifi network and bluetooth.

At that time on the device I see that it searches for android auto then it seems to boot but the message of android auto in search is not removed.

And on the console it does not start.

nisargjhaveri, 2024-01-17:
Try builds from #35 if they help by any chance?

rcazpeitia, 2024-01-17:
> Try builds from #35 if they help by any chance?

ok ok

From what I understand then first I feed the Pi device, I connect and connect to the Wifi network, after that I make the USB connection and then the BT connection?

Thank you very much in advance.

fkloft, 2024-01-17:
@rcazpeitia I'm using the default radio that ships with the Swift. You do not need to connect to the Pi's Wi-Fi; Android Auto will connect automatically and set up the phone so AA can use the Wifi exclusively (so the phone won't shut down mobile data).

fkloft, 2024-01-17:
@nisargjhaveri I've been using the RPi Zero W for a few days now and I'm happy to report that it works most of the time; I'm guessing because the Zero is slower to boot and the radio is more likely to be ready by the time boot is completed. However, it still happens occasionally that the radio won't connect and I have to reboot the Pi.
I'll try the build from #35.

rcazpeitia, 2024-01-17:
@fkloft 
Hello, 

Do you have any video of how you make the connection?

I have tried several ways but I can make it work.

Additionally how do you power the PI Zero W?

Do you just connect the power to the USB port?

Thanks in advance.

fkloft, 2024-01-17:
@rcazpeitia On the Pi Zero W, connect the left micro USB port to the car's radio (this one has power and USB OTG, the right port only is for power).
After the Pi has booted, connect your phone over Bluetooth to the Pi. After that, AA should just work. You might have to reboot the Pi by unplugging it. Wi-Fi is handled by AA.

nisargjhaveri, 2024-09-19:
The original issue seems to have resolved. Also there have been many changes incorporated in the latest build.

Closing this issue for now.

closed 2024-09-19 by nisargjhaveri