nisargjhaveri/WirelessAndroidAutoDongle #32: No bluetooth with Raspberry Pi zero 2w

xyzPrimeTV, 2024-01-08:
Hello @nisargjhaveri , 
First of all, a big thank you for this much needed project.

I am using the **v0.2.0** with Raspberry Pi zero 2w on my **TATA Nexon (XZA+) car**.
And unfortunately, no Bluetooth is getting listed to connect.
Wi-FI is visible.

It was working fine earlier. But from last month or two approximately, it stopped working.
In this time-frame, I think that my car's headunit  got some update in one of the servicing.
So I am suspecting that after this update it stopped working.

I can provide you logs but I need some help to generate the logs.
I am able to connect with SSH but I don't know where the logs are generated and if there is any specific command line to hit in shell ?

Thank you for your help in advance.

xyzPrimeTV, 2024-01-11:
@hkfuertes can you help me to check the errors ? Where I can check the errors so I can try to fix it ? I don't see much information on debugging part.
Thanks in advance.

nisargjhaveri, 2024-01-11:
> I am able to connect with SSH but I don't know where the logs are generated and if there is any specific command line to hit in shell ?

You can have a look at /var/log/messages file, it should have most relevant information. You can also share the file here if possible to have more eyes.

nisargjhaveri, 2024-01-13:
You can maybe also try the builds from #35, [builds here](https://github.com/nisargjhaveri/AAWirelessDongle/actions/runs/7506574825), and see if it helps?

xyzPrimeTV, 2024-01-16:
@nisargjhaveri , I tried with the mentioned build and now Bluetooth gets connected and wifi also but it still shows the message "Searching for android auto...". So I am still not able to connect the android successfully.
I also checked the rootfs/var/log folder but it is empty.

nisargjhaveri, 2024-01-17:
Thanks for trying out. Unfortunately the logs are only available when the device is on, they're not persisted. See [the troubleshoot section in the README](https://github.com/nisargjhaveri/AAWirelessDongle?tab=readme-ov-file#troubleshoot) in case it helps.

nisargjhaveri, 2024-09-19:
Closing as the original issue was resolved. Feel free to create a new issue of you're still facing any issue with the latest version.

closed 2024-09-19 by nisargjhaveri