nisargjhaveri/WirelessAndroidAutoDongle #33: Not working with VW Vento 2020 model OEM Head Unit

FAROKHDUBASH, 2024-01-09:
Thanks for this awesome project it works flawlessly in my Hyundai car, however it does not seem to work in my VW Vento, it powers on but it is not discoverable in the phone's BT settings. I was able to get it paired only once & the head unit showed AA, Miracast or carplay but the wireless AA didn't launch phone keeps showing searching for AA.

I tried to troubleshoot by unpairing it & I am unable to get it working.

IF you can help me out with how to debug I can perform it & provide you with the logs for troubleshooting.

nisargjhaveri, 2024-01-11:
Please have a look at https://github.com/nisargjhaveri/AAWirelessDongle/issues/30#issuecomment-1870512835, it has some info on where to start troubleshooting.

nisargjhaveri, 2024-01-13:
@FAROKHDUBASH You can try the builds from #35, [builds here](https://github.com/nisargjhaveri/AAWirelessDongle/actions/runs/7506574825), and see if it helps?

nisargjhaveri, 2024-02-25:
From the logs, looks like it connects fine to the phone and enabled the usb correctly. But the headunit is not starting the usb communication after that. Not really sure why is that.

Does the RPi boots up much before the head unit is started in your car?

FAROKHDUBASH, 2024-02-28:
The RPi takes about 10 seconds to boot up to the login screen.
I have tried by connecting the RPi to the car before starting the ignition & as well as starting the car & then plugging in the RPi.

If there is any specific steps that I should follow & collect the logs, please do let me know, I can collect them for you.

FAROKHDUBASH, 2024-03-20:
Hi Nisarg,

So I saw that 0.3.1 build was uploaded so I decided to give it a try, below are my observations.

I was able to get it to kind of work

1. Connect my laptop to the AAWirelessDongle Wifi access point
2. Pair my phone with the AAWirelessDongle via bluetooth

The above sequence seemed to work & I got many success attempts, however every time that I switched off the dongle & tried to reconnect it, the car head unit would not display the wireless AA interface.

I had to select forget the bluetooth devices (Car HeadUnit name + AAWirelessDongle) & repeat steps 1 & 2 above for a successful connection.

I have also captured the logs of the successful attempts that could help you to implement the necessary changes.

Also uploaded the video of the process that I followed.

[Working logs.zip](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14665931/Working.logs.zip)

https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/assets/129649876/83f0b8f9-b222-4b1d-a85f-2fe758ce9d41

nisargjhaveri, 2024-03-20:
Would it be possible to get the logs when it wasn't able to connect? It might help. Also, you can experiment with builds from #72 in case it helps in your case.

FAROKHDUBASH, 2024-03-29:
Hi Nisarg, Here you go, I have captured a working log & 4 attempts of a not working log for your reference.

[Working logs.zip](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14800971/Working.logs.zip)
[Not Working logs.zip](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14800980/Not.Working.logs.zip)

nisargjhaveri, 2024-03-30:
Please try the latest builds from main or try out latest build from #72. There have been new changes to handle some of the failures and retry in those cases, might help in your case.

Not very clear from the logs, but when it fails, the headunit does not seem to respond after the USB is enabled. Might be a timing issue. Also, it really shouldn't be required to repiar the phone via Bluetooth every time.

FAROKHDUBASH, 2024-04-01:
Thanks Nisarg, Tried Build 0.4.0, along with customized AAWG variable of 1 that seems to work somewhat consistently with newer phones (14.1), does not work as intended on older phones (10 & 11) (that should be ok I guess).

With AAWG variable set to 0, I had no success, even the previous method of first connecting the laptop to wifi and then pairing from phone did not seem to work.

With AAWG variable set to 2, I can just connect to the wifi but the BT does not show up to pair in the phone.

I am attaching some logs for your reference for further product enhancements. (if the folder is empty, means I was unsuccessful in connection method so no logs collected.)

[1 Apr 2024 Working.zip](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14822109/1.Apr.2024.Working.zip)

[1 Apr 2024 Not Working.zip](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14822108/1.Apr.2024.Not.Working.zip)

So to summarize I am currently using a newer phone (Android 14.1) with build 0.4.0 with AAWG variable set to 1, which works 6 out of 10 times, which is usable for me until a newer release that can fix the reliability. Thanks once again.

nisargjhaveri, 2024-04-01:
With dongle mode (`AAWG_CONNECTION_STRATEGY=0`), you first need to connect he phone via usb. And when trying to connect to the dongle, the connection only starts once the phone is connected to headunit also via Bluetooth. Is this happening in your case?

From the logs, I have two hypothesis:
1. The headunit has a very specific time period, only in which it can detect and start Android Auto via USB.
2. Or the head unit is simply not able to detect that usb is enabled from dongle for some reason in some cases.

prashantchoudhary, 2024-08-18:
Used it with 2022 VW vento. Worked fine. I used the prebuilt image for Raspberry Pi Zero 2W. Also the micro usb input goes into data port and not the power port(just in case it helps).