nisargjhaveri/WirelessAndroidAutoDongle #37: Issues with harware

VanshRIT, 2024-01-17:
Thank you for your hard work on the project. I drive a Mazda 3 2023 with wired Android Auto. I tried to implement your solution, and it worked successfully on the Raspberry Pi 4 8GB board. However, I also attempted to use it on a Raspberry Pi Zero, but I was unsuccessful.

nisargjhaveri, 2024-01-17:
There can be couple of differences I can think of. The time to boot could make a difference in case the headunit has some tricky timeouts. Apart from that, I don't see much reason on why this out happen. Any chance can you get some logs out from both successful and unsuccessful attempts and compare?

chuppito, 2024-02-18:
Hello, this project is very interesting. Do you confirm that the raspberry pi 0 2w now works?
What material do you think will be the least expensive to make wireless AA? THANKS

kipperfish, 2024-02-23:
I have a Raspberry Pi Zero 2 W and have just tried both the release image and the latest CI build from Main. Unfortunately both fail to boot (I tried the Raspberry Pi OS Lite image which worked on the same Pi, with the same card). All I get is two flashes of the green light and then nothing.
Does anyone have any ideas?

nisargjhaveri, 2024-03-01:
Not sure about failing to boot completely, but we've updated some firmwares via 45cf917d557d7abb847d627ffd180da5319ffbfc. Might be worth to try out the latest main builds.

kipperfish, 2024-03-01:
I just tried the latest build available on main that includes that commit
(from about 13 hours ago, one more was just kick off)

I get the same behaviour as with v0.3.
I can pair to the dongle, and it shows up on the cars head unit, but the
phone then sits on "Looking for Andoid Auto"

Logs attached

Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-03-01 07:28:45 UTC)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61 ***@***.***) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 Fri Mar  1 08:00:43 UTC 2024
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:38:02:62 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 340196K/421888K available (7906K kernel code, 702K rwdata, 2184K rodata, 256K init, 498K bss, 16156K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000006] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000057] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000158] bcm2835: system timer (irq = 27)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000947] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001445] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001524] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060289] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060394] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060586] LSM: Security Framework initializing
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060798] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060869] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.062269] cgroup: Disabling memory control group subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064164] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064246] cblist_init_generic: Setting shift to 0 and lim to 1.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064568] Setting up static identity map for 0x8220 - 0x8258
Jan  1 00:00:03 buildroot kern.info kernel: [    0.065822] devtmpfs: initialized
[... cut ...]
Jan  1 00:01:16 buildroot kern.info kernel: [   76.948330] dwc2 20980000.usb: bound driver configfs-gadget.default
Jan  1 00:01:17 buildroot kern.info kernel: [   77.175874] dwc2 20980000.usb: new device is high-speed
Jan  1 00:01:17 buildroot kern.info kernel: [   77.355332] android_work: sent uevent USB_STATE=CONNECTED
Jan  1 00:01:17 buildroot kern.info kernel: [   77.357443] dwc2 20980000.usb: new address 1
Jan  1 00:01:17 buildroot kern.info kernel: [   77.381587] android_work: sent uevent USB_STATE=CONFIGURED

nisargjhaveri, 2024-04-01:
The log suggests that the car never starts the usb connection after we enable the usb gadget. One of the common reason has been faulty data cables. Make sure the cable works?

Also, please try the latest build, there have been a bunch of changes which might help in thi case.

In any case, it the issue still persists, please create a new issue. This one contains multiple different problems and makes the discussion very difficult.

closed 2024-04-01 by nisargjhaveri