nisargjhaveri/WirelessAndroidAutoDongle #38: It does not start android auto susuki jimny

rcazpeitia, 2024-01-17:
How are you doing?

What kind of radio do you use in the swift?

I have a jimny 2024 and by USB it detects without problems my cell phone, when I connect my Pi zero W, I connect the device to the wifi network and bluetooth.

At that time on the device I see that it searches for android auto then it seems to boot but the message of android auto in search is not removed.

And on the console it does not start.

recwil, 2024-01-18:
FWIW same issue with my 2017 Focus RS. Phone detects the pi, connects via bluetooth (which then pairs me with the car as well) and then appears to initialise Android Auto but then displays a "looking for/searching for" connection notification and then nothing.

recwil, 2024-01-18:
^ this issue was fixed for me by using the latest build generated here https://github.com/nisargjhaveri/AAWirelessDongle/pull/35
Obviously was a timeout issue with the Ford Sync 3 head unit.

rcazpeitia, 2024-01-18:
@recwil 

Right, I will download the latest version and try it.

I will notify if the process is solved.

Thank you all very much.

Thanks a lot.

cr08, 2024-02-13:
> ^ this issue was fixed for me by using the latest build generated here #35 Obviously was a timeout issue with the Ford Sync 3 head unit.

Just came across this repo and have Sync 3 in my Ford vehicle. Grabbed the latest [artifact](https://github.com/nisargjhaveri/AAWirelessDongle/actions/runs/7856371400) and ran it on my 0w. For me it still doesn't connect on the first try when the vehicle is started/Sync boots. Power cycling the Pi with the vehicle still running and it comes up fine. This is 100% repeatable across a number of drive cycles in my case.

My Sync system is also jailbroken and I have access to the internal logs if there's anything that may be useful there to help troubleshoot.

Phone: Pixel 4a (Stock Android 13)
Vehicle: Ford C-Max Energi
Headunit: Stock Ford Sync 3 (v3.4.23188)

nisargjhaveri, 2024-02-14:
@cr08 could you please create another issue as it doesn't seem related to the original issue here and the original issue is likely already fixed. Also, please share logs from the pi if possible (see Troubleshoot section in README).

nisargjhaveri, 2024-02-14:
@rcazpeitia closing this issue assuming it has been fixed in latest main. Feel free to reopen if this is still an issue.

closed 2024-02-14 by nisargjhaveri

rcazpeitia, 2024-02-14:
@nisargjhaveri 

Sorry for not replying, in the end I opted for a ready to run device and it worked correctly.

I think I was doing the steps wrong on my part, but I thank you for the comments and your work.

If I can today I will test again and report the result.

Regards.