nisargjhaveri/WirelessAndroidAutoDongle #39: First time connect Issue

DeadEnd0o7, 2024-01-20:
There is a need to reconnect the usb to get a connection where it gets stuck at searching for android auto but after that it works as expected. Tested on TATA Tiago.
Is there a possibility to do a similar setup for apple carplay?

Andreunto, 2024-01-21:
I tried to do this and it always stays on "search for android auto..."

Could it be because the Renault Medianav is based on Windows CE?

nisargjhaveri, 2024-01-21:
This is likely because of some timeout issue with headunit. Try out the generated builds from #35, might help.

Andreunto, 2024-01-21:
> This is likely because of some timeout issue with headunit. Try out the generated builds from #35, might help.

I've already tried that build but it still doesn't work

nisargjhaveri, 2024-01-21:
See if you can get some logs out? See Troubleshoot information in the README.

MrCrrispy, 2024-01-25:
I have the same issue on my Hyundai Ioniq EV and Pixel 7a

I have tried the other build but get stuck on "searching for Android auto".

Here is a copy of my /var/log/messages
`
# cat /var/log/messages
Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-01-25 12:43:39 CET)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (root@Soyuz) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot 2023.11-620-g7a59c3a4d4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Thu Jan 25 13:25:59 CET 2024
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:69:18:89 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]      Tracing variant of Tasks RCU enabled.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000018] Switching to timer-based delay loop, resolution 52ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000345] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000857] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000909] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000950] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001009] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001137] LSM: Security Framework initializing
[... cut ...]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:03 buildroot
Ett fel uppstod. Ditt meddelande skickades inte.
Show quoted text
`

MrCrrispy, 2024-01-27:
Could this be the main issue?

Jan 1 00:00:08 buildroot daemon.info rngd: [jitter]: Unable to obtain AES key, disabling JITTER source

nisargjhaveri, 2024-01-27:
I've seen rngd often complain about this. It generate works fine even this being logged. 

Something looks wrong here. The phone connects to the associates with the wifi but gets disassociated in around 4 seconds. And the WifiConnectStatus sees to be some error (length not equal to 2 generally indicates some error). 

Try connecting to the WiFi directly? Does the phone connect okay with the raspi wifi? Do you have vpn connected when this happens? 

```
Jan 1 00:00:13 buildroot user.info aawgd[261]: Read WifiStartResponse. length: 2, messageId: 7
Jan 1 00:00:28 buildroot daemon.info hostapd: wlan0: STA 3e:7f:7f:b9:d9:fd IEEE 802.11: associated
Jan 1 00:00:32 buildroot daemon.info hostapd: wlan0: STA 3e:7f:7f:b9:d9:fd IEEE 802.11: disassociated
Jan 1 00:00:32 buildroot user.info aawgd[261]: Read WifiConnectStatus. length: 11, messageId: 6
```

MrCrrispy, 2024-01-27:
I am connecting to the dongle wifi from my phone to get the logs, so that works fine.

I use VPN, and I have tested with VPN On and VPN off, none of them work.

I did another test today and I got a popup on the phone saying something like "Continue using Android auto on the car screen". Then after a few seconds on my car screen I get a message saying "Phone is not responding"

Saved the log but it looks very similar I think.

<details open>
summary>Log here</summary>
cat /var/log/messages
Jan  1 00:00:03 buildroot [syslog.info](http://syslog.info/) syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-01-25 12:43:39 CET)
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (root@Soyuz) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot 2023.11-620-g7a59c3a4d4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Thu Jan 25 13:25:59 CET 2024
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000]   Normal   empty
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:69:18:89 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000]      Tracing variant of Tasks RCU enabled.
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Jan  1 00:00:03 buildroot [kern.info](http://kern.info/) kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[... cut ...]
Jan  1 00:00:55 buildroot [daemon.info](http://daemon.info/) dnsmasq-dhcp[225]: DHCPACK(wlan0) 10.0.0.6 16:b4:a2:c8:d2:0e Henrik-Pixel-7a
Jan  1 00:01:04 buildroot [authpriv.info](http://authpriv.info/) dropbear[268]: Child connection from [10.0.0.6:50366](http://10.0.0.6:50366/)
Jan  1 00:01:04 buildroot authpriv.notice dropbear[268]: Password auth succeeded for 'root' from [10.0.0.6:50366](http://10.0.0.6:50366/)
Jan  1 00:01:07 buildroot [daemon.info](http://daemon.info/) hostapd: wlan0: STA 3e:7d:18:a7:f4:36 IEEE 802.11: disassociated

</details>

nisargjhaveri, 2024-01-30:
Okay, next would be to have verbose logging for hostapd. You can try passing `-dd` from the init file you'd have in /etc/init.d/ for hostapd?

nisargjhaveri, 2024-02-06:
This is very strange, this time the communication did start and then stopped and Wifi disconnected some time later. The last log looks like USB might've stopped working. Please check the cable, seems like a common issue? Though the earlier logs were different.

MrCrrispy, 2024-02-09:
I tested 9 different usb cables, and checked that they work on my PC before but no improvement.

I have not been able to get the communication to start as it did in the previous log, not sure what I did different that time.

I also tested with a another phone (Samsung A40) but no success.
However with the Samsung I first paired bluetooth and it then switched to wifi, but the phone was saying "wrong password" and would not connect to wifi. So I connected to wifi manually first and entered the password, and then connected to bluetooth again and it was able to connect like on my Pixel, but then getting disassociated like before.

I was expecting that the wifi password would be shared over bluetooth and that I would not need to enter it manually, is that not the way it is supposed to be?

MrCrrispy, 2024-02-09:
I did some file searching and discovered that the wifi password was also set in the common.cpp file, so I updated that as well, I also tested using the default passwords everywhere. But still no improvement.

nisargjhaveri, 2024-02-11:
If you're trying to use a different password than the default one you can have a look at https://github.com/nisargjhaveri/AAWirelessDongle/issues/42.

Changing root password should not be an issue. 

You should not need to specify the password on phone. The AA connection protocol should take care of it.

MrCrrispy, 2024-02-11:
So...turns out my issue was purely incompetens on my part.
I was still on the main branch and had not switched to the "optimistic phone connection" branch as I thought I did.

So after switching everything now works perfectly!

Big thanks for your help @nisargjhaveri, sorry for wasting your time.

nisargjhaveri, 2024-02-11:
Glad that you were able to make it work. The "optimistic phone connection" changes are now merged in main branch. The lastest builds from main should also probably work for you in that case.

chanakyaraichur, 2024-02-13:
@nisargjhaveri  @MrCrrispy  Can you provide image for raspberrypi0w with this fix ? 
I am facing issues in making the build .

borsosgy, 2024-02-13:
Hi nisargjhaveri,
First of all I would like to say a big thank you for your efforts.

I had the same problem as MrCrrispy.
My LOG was almost the same, I also tested lots of USB cables.
The problem was the same, the head unit timing was bad. After starting the AAWirelessDongle, the Bluetooth connection was not established so quickly and the head unit reported errors.
I made the new build (https://github.com/nisargjhaveri/AAWirelessDongle/pull/35 "optimistic phone connection") on Pi Zero 1.1.(pi0w)
The "optimistic phone connection" function works, but now the Bluetooth communication fails and does not return.
This means that AAWirelessDogle is paired with the cell phone via Bluetooth, --> Android Auto starts automatically on the Headunit, --> the Bluetooth connection between AAWirelessDongle and Phone fails, -->Android Auto started works, but without Bluetooth connection, so the sound doesn't work on the hay unit.

Do you have an idea what I could do?

thanks in advance

Gyula

borsosgy, 2024-02-13:
I made the new image from the source:
git clone --recurse-submodules https://github.com/nisargjhaveri/AAWirelessDongle
I think that should be good, right?

nisargjhaveri, 2024-02-14:
@borsosgy the source looks correct, you can see the instructions in the README as well. Also, could you please create another issue on calls not working over bluetooth? It seems different than the original issue here.

nisargjhaveri, 2024-02-14:
@chanakyaraichur you can download the latest artefacts from recent builds for main branch.
https://github.com/nisargjhaveri/AAWirelessDongle/actions?query=branch%3Amain

nisargjhaveri, 2024-02-14:
I'll close this issue as it had digressed from the original report. Feel free to create another issue or reopen if it is very similar.

closed 2024-02-14 by nisargjhaveri