nisargjhaveri/WirelessAndroidAutoDongle #41: Change DPI

brianmagicr, 2024-01-23:
Would it be possible to implement a way to change the dpi on the headunit?

nisargjhaveri, 2024-01-24:
I believe this will require us to decode and act as man-in-the-middle for the AA communication. We currently simply pass through all communication as is. While not impossible, this would certainly add much more complexity.

Kraken1337, 2025-01-25:
You can change the resolution in the development settings in android auto

un4saken, 2025-03-05:
> You can change the resolution in the development settings in android auto

It doesn't make any difference. AAwireless has that option to change DPI so you can make the screen smaller and see more icons , map areas, report button in google maps etc.

manio, 2025-03-27:
@un4saken I implemented experimental MITM mode in [aa-proxy-rs](https://github.com/manio/aa-proxy-rs) and I am quite interested in this feature... Can you tell me more about how it works exactly in the AAwireless? Maybe you have some screenshot what options are available?

manio, 2025-03-29:
OK, I am able to lower the `density` setting which makes UI smaller, so for the first time I can see a `Report` button in Google Maps on my small (800x480) car display :)

Kraken1337, 2025-03-29:
> [@un4saken](https://github.com/un4saken) I implemented experimental MITM mode in [aa-proxy-rs](https://github.com/manio/aa-proxy-rs) and I am quite interested in this feature... Can you tell me more [...]

How did you do that?

manio, 2025-03-30:
I just changed the `MESSAGE_SERVICE_DISCOVERY_RESPONSE` which my car is sending to the phone. There is hardcoded "dpi" value of 160. I changed it to 130.
And here is how it looks before and after:

[image]

[image]

I'll soon add `--dpi` parameter in my daemon so everyone can use it...

manio, 2025-03-31:
FWIW, if you're interested, this is now done via `--dpi` parameter in `aa-proxy-rs` (at least it's working in my Renault HU):
https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#dpi-settings

Kraken1337, 2025-03-31:
> FWIW, if you're interested, this is now done via `--dpi` parameter in `aa-proxy-rs` (at least it's working in my Renault HU): https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#dpi-settings

nice, i will try it soon

Kraken1337, 2025-04-16:
> FWIW, if you're interested, this is now done via `--dpi` parameter in `aa-proxy-rs` (at least it's working in my Renault HU): https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#dpi-settings

Do you mind building a new release by any chance? :D

manio, 2025-04-16:
> Do you mind building a new release by any chance? :D

OK, will try to find some time (I have to adjust it to recent Nisarg's changes).

manio, 2025-04-18:
@Kraken1337
It's now [released](https://github.com/manio/aa-proxy-rs/releases).