nisargjhaveri/WirelessAndroidAutoDongle #43: Using for Tata Nexon 2023 edition - Audio Music Lag when using touchscreen.

mddanish, 2024-01-26:
Welldone for creating a very usefull and money saving project.
I'm using it with my RPi0-W, connect very nicesly and good.

The only problem I noticed when Music is playing & streaming to car infotainment, and when using Car touchscreen to navigate or use map to set, Audio starts lagging.

Ioniq3, 2024-02-03:
Hello mddanish,

If this help you, I have the same RPI0W and after disable the hdmi output the performance increases, so the hdmi drains a lot of energy of the board.
It is done add the following line in the config.txt 

hdmi_blanking=1

mddanish, 2024-02-04:
Thanks Mate, it works

gamelaster, 2024-02-06:
@nisargjhaveri wouldn't be good to disable hdmi by default?

nisargjhaveri, 2024-02-11:
Should be fixed by #48.

closed 2024-02-11 by nisargjhaveri