nisargjhaveri/WirelessAndroidAutoDongle #44: Feature request: audio routing

ansher, 2024-01-27:
Hello great community.

I would like to propose the configuration where audio sync stays with the phone vs being sent to the headunit via AAWirelessDongle.

The value of this configuration is a choice user can make whether audio is routed via BT of the phone connected to a head unit directly or via AAWirelessDongle, because in some cars there are differences in audio quality.

This feature has been implemented by Emil B. in AAwireless and I think it's amazing, I can hear a great degree of difference in my car.

Thank you !

spotopolis, 2024-09-02:
I would like this function as well. My aftermarket head unit supports wireless AA but audio quality is bad and dull compared to if I route audio though Bluetooth. The downside is I am not able to route audio through Bluetooth on my phone while using wireless android auto. It will automatically reconnect audio though wifi and force me to use it that way, so I've only used audio with Bluetooth and have not used android auto at all. If I am able to use this function via rpi, that would be fantastic.

sbfox85, 2024-10-06:
I would love this, especially if Audio could be routed to a DAC on the raspberry pi, it would allow me to use a decent amplifier but keep my original head unit.

Global-Citizen, 2025-02-21:
Yes, I also would like to have this feature. It's called "disable audio sink" in the AAwireless.

manio, 2025-07-19:
FWIW: https://github.com/aa-proxy/aa-proxy-rs?tab=readme-ov-file#features