nisargjhaveri/WirelessAndroidAutoDongle #45: Access to other services in the raspberry pi

Ioniq3, 2024-02-03:
Hello everyone,

I don't know if it is a characteristic of the android auto protocol or not. Sorry if the answer is yes.
Could be possible to access to other ports apart from the tcp server for android auto at the same time android auto is working.
I tried but it doesn't work, for example access to ssh server from the mobile phone, but i don't know if it is a issue of the code o the protocol itself.

For example to run a subsonic server in the rapsberry in the 10.0.0.1 at some port, so the mobile can connect wireless to the RPI for android auto and to the server to allow symfonium or other app subsonic compatible read and no used mobile data for music. Another posibility is to access samba despiste there is few music broser for samba in google play.

nisargjhaveri, 2024-02-06:
When Android Auto is connected, it connects to the Wifi in local only mode. Generally the Wifi network is only available to Android Auto, AFAIK. I don't really know if there is a workaround to allow other apps to connect using the connected Wifi network.

Ioniq3, 2024-02-08:
Thanks, I think it is not possible in android auto protocol. I will investigate to force the raspberry to act as a hostpad and wifi client to connect to share conection in the phone, so the phone can connect to rpi throught the second IP of the mobile hotspot. For what i search it will be possible to the rpi act as AP server and Wifi Client at the same time for routed configurations

closed 2024-02-08 by Ioniq3