nisargjhaveri/WirelessAndroidAutoDongle #46: My Car has an Inbuit Internet Wifi

mddanish, 2024-02-04:
What changes should I look into, so that rather than create a new wifi, Device connect to my car inbuilt WIFI, and my phone also connected to my car wifi.

nisargjhaveri, 2024-03-01:
When phone connects to Android Auto, it generally uses local only connection, which means that even if it connects to the car's wifi, it won't use it for the internet.

Having said that, it should be possible to tweak the configuration a bit to allow the dongle in client mode:
- Disable hostapd and enable wpa_supplicant in /etc/network/interfaces
- Update the /etc/wpa_supplicant.conf file to include wifi credentials
- Set `AAWG_WIFI_SSID`, `AAWG_WIFI_PASSWORD`, `AAWG_WIFI_BSSID` and `AAWG_PROXY_IP_ADDRESS` env variables appropriately for aawgd.

godoppl, 2024-04-15:
I have the same idea in mind. I have done a little testing, and have not been successful.

I have updated network and wpa config to connect to the WiFi. This works, as I can SSH into the device on that network
I have also edited the environment file `/etc/aawgd.env` and added the appropriate environment variables, but nothing happens.

However I am a bit uncertain about the what the proxy IP should point to, I have set it to the gateway address.
Also I am not connecting to the BT network, since that forces the phone to disconnect from the cars WiFi.

I was hoping to not have to modify and build the project myself, but if I have to, I will :)
I hope this can be achieved from configuration alone. Any ideas?

tonydee2377, 2024-04-19:
I'll answer myself.
After a few tests it works.
I installed v0.3.1 first and it worked after a few minutes of waiting, then I reformatted the SD card well and installed the latest version v0.4.0 and this also worked.
I don't know what happened before, but having a few USB cables around, I found that many were fake. Now I continued to make customizations one at a time, until I discovered that changing the SSID and password no longer made anything work, but I also solved this by putting the right variables in the /etc/aawgd.env file, also tampering with the bluetooth file, but now everything works fine, I get into the car in time to turn on the navigator and Android auto connected via wifi.

Now having a 32Gb SD card and therefore some free space, I created a FAT32 partition on the remaining space in order to put mp3s and videos and use the zero rpi w also as an mp3 USB stick.

But it doesn't work, can anyone tell me how to show this partition on the USB cable to the car radio?
I understand that the USB is busy making the phone's Android Auto communicate with the car, but if I could switch to also use the space as an archive for my playlists, it wouldn't be bad.

I will try these days, in the meantime if anyone can help me I would be grateful.

nisargjhaveri, 2024-04-21:
> However I am a bit uncertain about the what the proxy IP should point to, I have set it to the gateway address.

The proxy IP should point to the address assigned to the dongle from your Wifi, not the gateway address.

> Also I am not connecting to the BT network, since that forces the phone to disconnect from the cars WiFi.

Once the config is correct, once you connect to the BT, phone should connect to the same Wifi again. It might connect as a local only wifi though as mentioned before in this thread.