nisargjhaveri/WirelessAndroidAutoDongle #47: AAWireless connection Developer Mode

szafir51, 2024-02-06:
Hello nisargjhaveri,

First I would like to thank you for your great work and let you know that your software installed on Rpi0w works great with Renault Megane 4 '2019 (R-Link2 media system).

But there is one problem that is not related to your application, but can maybe it be solved with it.
Since Android 14, Android Auto does not display apps installed outside of the Google Play Store. Third-party Apps like Fermata Auto and Screen2Auto are no longer shown on the Android Auto screen.
More information here:
https://github.com/AndreyPavlenko/Fermata/issues/327

The only way to run applications installed from outside the Play Store is the original AAWireless.io adapter with the Developer Mode option enabled in their application: 
https://play.google.com/store/apps/details?id=app.aawireless
https://i.redd.it/aawireless-settings-for-kia-hyundai-genesis-v0-braceaqsutmb1.png?s=5b566795c38d799e4a2b352b9d139cfbf634f4e3

Is it possible to enable a similar connection mode in your application?

szafir51, 2024-02-06:
Probably you are right. I compiled the software, only changed the Bluetooth name to AAWireless-pQibvCgx and the AAWireless.io software allowed the application to run but nothing else could be done. Pity.

chuppito, 2024-02-18:
Hello, aawireless uses a third party application as an emulator. This may be why it works on the dongle and not the raspberry. For Fermata, it works with head unit reload and can be AAgateway (this is the application that allowed Emil to do the aawireless project). For it to work on the raspberry, I suppose you have to create an application that communicates with the phone in developer mode?

tiliarou, 2024-03-30:
Interested in this to see how aawireless dongle/app do it

Ioniq3, 2024-04-01:
Hi, you can activate developer mode in android auto press several time the version in the android auto app in your mobile phone

szafir51, 2024-04-02:
Not in Android 14. If you want to use third-party app in Android 14 AA, you need to use AAWireless.io adapter with their app AAWireless for Android Auto. You need to enable Developer mode in this app.

tiliarou, 2024-04-07:
In AA 14 on my samsung S21 ultra, pressing the build number several times activates de developper mode. But you mean that it's not enough to see other 3rd party apps ? Why would some appear but not others ?
So the AAwireless.io app developer mode option is doing something differently than activating the option in android auto app ?

Ioniq3, 2024-04-08:
> In AA 14 on my samsung S21 ultra, pressing the build number several times activates de developper mode. But you mean that it's not enough to see other 3rd party apps ? Why would some appear but not [...]

You have to activate the developper mode in the android auto app, and then activate allow unknown sources in the developper settings in the 3 point icon of the android auto app

chuppito, 2024-04-08:
Hello, @tiliarou  Means there is a difference between this application and aa.wireless

With the phone's developer mode, some modified applications do not work. Before Android 14, we installed with kinginstaller. Since then you either have to be root or buy the aawireless dongle.
Users of this app would like to be able to do the same thing I guess

tiliarou, 2024-04-08:
> Hello, @tiliarou Means there is a difference between this application and aa.wireless
>
> With the phone's developer mode, some modified applications do not work. Before Android 14, we installed with [...]

Yes I understand, and that's my goal as well to be able to do like aawireless dongle but with my pi 0 2w.
My question was to ask if being able to enable developper mode by pressing build 7 times in android auto app on the phone (which I've been able to do) is the same thing as enabling developper mode in the aawireless dongle app.
My understanding is that it's not the case but I would like to understand why and what is done differently.

szafir51, 2024-04-08:
Because AAWireless.io app works like men-in-the-middle app. It is not only redirecting signal from USB cable connection to Wifi/BT like other adapters.

tiliarou, 2024-04-15:
I found an old forked repo from the dev (I suppose) of AAwireless dongle, if that might be useful : https://github.com/chutchi2/aawireless

pgolos21, 2024-08-25:
> Author

Do you still have the file you have made that make it open in the app but for the raspberry pi  zero w

Tibingeo, 2024-09-29:
Any luck with this on Android 14 ? 
I am using Android 14, using this project connecting wireless. I would like to know how we can stream videos.

DamianoP, 2025-08-09:
This feature request is fantastic, are there any updates?

manio, 2025-08-09:
> This feature request is fantastic, are there any updates?

Yes, I implemented it in my project:
https://github.com/aa-proxy/aa-proxy-rs?tab=readme-ov-file#features

DamianoP, 2025-08-09:
> > This feature request is fantastic, are there any updates?
> 
> Yes, I implemented it in my project: https://github.com/aa-proxy/aa-proxy-rs?tab=readme-ov-file#features

Wow, congratulations!