nisargjhaveri/WirelessAndroidAutoDongle #51: Question: Compatibility of AAWireless with Pi OS and CSI camera setup

JazzzzX, 2024-02-12:
Hello,

I'm interested in learning whether the AAWirelessDongle is compatible with a standard Raspberry Pi OS image. If so, are there specific installation procedures or packages required? I currently have a Raspberry Pi with a CSI camera functioning as a simple dashcam. My ultimate goal is to configure my Pi to serve as both an AAWireless dongle and a dashcam simultaneously.

As my current image lacks appropriate tools like apt or a package manager for CSI camera configuration, I'm seeking guidance on how to proceed with integrating both functionalities.

nisargjhaveri, 2024-02-17:
We do need some kernel modifications for the android accessory driver to work. This would makes it difficult to get it working on the stock Raspberry Pi OS.

JazzzzX, 2024-02-17:
I can understand, Thanks

closed 2024-02-17 by JazzzzX