nisargjhaveri/WirelessAndroidAutoDongle #52: Release binary images

GreenZeeGuy, 2024-02-12:
Hello AAWirelessDongle team,

It looks like there has no image released since Nov 2023 even though many issues have been fixed and checked in.
It will be great to have an updated set of binaries to try out !

Thanks in advance !

nisargjhaveri, 2024-02-13:
Thanks for reminding. I was waiting for some of the changes to stabilize before creating a release, but will do this soon.

In the meantime you can use the generated builds from one of the recent CI runs on main branch.
https://github.com/nisargjhaveri/AAWirelessDongle/actions?query=branch%3Amain

nisargjhaveri, 2024-02-23:
A new release has been created: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/releases/tag/v0.3.0

closed 2024-02-23 by nisargjhaveri