nisargjhaveri/WirelessAndroidAutoDongle #54: bluetooth dropout while connected

borsosgy, 2024-02-14:
I made the new build (https://github.com/nisargjhaveri/AAWirelessDongle/pull/35 "optimistic phone connection") on Pi Zero 1.1.(pi0w)
The "optimistic phone connection" function works, but now the Bluetooth communication fails and does not return.
This means that AAWirelessDogle is paired with the cell phone via Bluetooth, --> Android Auto starts automatically on the Headunit, --> the Bluetooth connection between AAWirelessDongle and Phone fails, -->Android Auto started and works on Headunit, but without Bluetooth connection, so the sound doesn't work on the Headunit.

Do you have an idea what I could do?

thanks in advance

Gyula

borsosgy, 2024-02-14:
[image]

borsosgy, 2024-02-14:
[image]

nisargjhaveri, 2024-02-15:
We disconnect and turn of Pi's bluetooth once the phone is connected to allow phone to directly use headunit's bluetooth for phone audio. Media audio still goes through pi using the wifi connection.

Once the phone is connected to the headunit, it is supposed to connect to the headunit bluetooth directly. Is this not happening?

There is a so called "dongle mode" for connection which the other similar adapters are implementing, which could help with this. Though, I haven't been able to figure out the details of this mode to implement yet.

borsosgy, 2024-02-18:
Please excuse my late reply.
It works fine, I misunderstood it.
It works like this through a USB cable, and I thought that's how it should be.
Everything was great with the last build. Thank you for your work.

nisargjhaveri, 2024-02-20:
Closing this issue since you were able to make it work. :)

closed 2024-02-20 by nisargjhaveri