nisargjhaveri/WirelessAndroidAutoDongle #55: Ford Sync 3 no connection on cold boot/first vehicle start

cr08, 2024-02-14:
As requested in the other issue, I'm adding this here along with my log files.

Came across this repo a few days ago and decided to give it a go. Went straight to the latest artifact as of the time of writing: https://github.com/nisargjhaveri/AAWirelessDongle/actions/runs/7856371400

This is on a 0w (not a '2' model) and connecting to a stock Pixel 4a with Android 13.

Leaving the Pi plugged in to the vehicle before powering up, when the vehicle/headunit starts up, there's no visible connection on the headunit. The phone is ready and has the telltale 'looking for AA connection' and actively connected to the Pi's SSID. Not until I powercycle the Pi with the vehicle running does it come up and seems to do so pretty reliably.

Attached my messages log file from that attempt:
[messages.zip](https://github.com/nisargjhaveri/AAWirelessDongle/files/14289227/messages.zip)

I also have log files from the headunit itself if needed, although they are a bit verbose and I need to redact some stuff like the GPS location that appears in numerous spots. Skimming over it I don't see anything obvious other than a 'not supported' USB message every 10-15 seconds or so. But I'm also not sure what I'd be looking for.

Site note during this most recent test and I can move this to its own issue if you'd like: I tried to use the assistant and it didn't seem to want to pick up my voice. The little animation at the bottom of the screen just stayed until it timed out. Using the native wired connection I've had zero issues with this. A quick search here and I don't readily see anything mentioned about this (unless I'm using the wrong search terms).

admiralspeedy, 2024-02-15:
My Pi 0 W is coming tomorrow, I will see if I observe the same behaviour in my 2019 Escape with Sync 3.4.23188 as well.

Cr4z33, 2024-02-15:
Hmm was going to try out my Pi with a Ford Ka+ Active, but now reading this I dunno...

admiralspeedy, 2024-02-15:
Can confirm, Same issue with my Escape running Sync v3.4.23188.

It also doesn't re-connect if you walk away from the car while it's running and then come back.

nisargjhaveri, 2024-02-16:
Looks like some exception while connecting usb. 

Is this an issue only with Pi 0w or other models as well?

```
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.869054] ------------[ cut here ]------------
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.871552] WARNING: CPU: 0 PID: 0 at drivers/usb/dwc2/gadget.c:3127 dwc2_hsotg_epint+0xe84/0xf90 [dwc2]
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.876889] Modules linked in: rfcomm usb_f_fs usb_f_accessory ipv6 cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg libcomposite dwc2 roles hci_uart btbcm bluetooth ecdh_generic ecc libaes brcmfmac brcmutil sha2
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.885519] CPU: 0 PID: 0 Comm: swapper Not tainted 6.1.61 #1
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.888467] Hardware name: BCM2835
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.891312]  unwind_backtrace from show_stack+0x10/0x14
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.894192]  show_stack from dump_stack_lvl+0x24/0x3c
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.897072]  dump_stack_lvl from __warn+0x78/0xc0
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.899964]  __warn from warn_slowpath_fmt+0x6c/0xc8
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.902908]  warn_slowpath_fmt from dwc2_hsotg_epint+0xe84/0xf90 [dwc2]
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.906177]  dwc2_hsotg_epint [dwc2] from dwc2_hsotg_irq+0x880/0xc40 [dwc2]
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.909623]  dwc2_hsotg_irq [dwc2] from __handle_irq_event_percpu+0x44/0x134
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.913011]  __handle_irq_event_percpu from handle_irq_event+0x28/0x68
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.916194]  handle_irq_event from handle_level_irq+0x9c/0x170
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.919430]  handle_level_irq from generic_handle_domain_irq+0x24/0x34
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.922775]  generic_handle_domain_irq from bcm2835_handle_irq+0x20/0x28
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.926204]  bcm2835_handle_irq from generic_handle_arch_irq+0x28/0x3c
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.929582]  generic_handle_arch_irq from __irq_svc+0x70/0x90
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.932883] Exception stack(0xc0a55f30 to 0xc0a55f78)
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.936208] 5f20:                                     00000000 00000000 00000001 00000000
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.942824] 5f40: c0a5a080 00000000 c0a570a0 d9bffc00 ffffffff 00000000 c0a42a64 00000000
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.949619] 5f60: 0020c49c c0a55f80 c000f298 c000f29c 60000013 ffffffff
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.953083]  __irq_svc from arch_cpu_idle+0x28/0x30
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.956522]  arch_cpu_idle from do_idle+0xb4/0xe8
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.959873]  do_idle from cpu_startup_entry+0x1c/0x20
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.963143]  cpu_startup_entry from rest_init+0x78/0x94
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.966323]  rest_init from arch_post_acpi_subsys_init+0x0/0x8
Jan  1 00:00:20 buildroot kern.warn kernel: [   20.969538] ---[ end trace 0000000000000000 ]---
```

cr08, 2024-02-16:
Only tested with the 0w so far here. I need to see what full size Pi models I have on hand I can try to test with. I also live near a Microcenter so I can try picking up a 02w later and test with that.

admiralspeedy, 2024-02-16:
I built an image for my Pi 0 W with the pull request for reconnecting last night, will report if it works when I get in my car after work today.

I do have a Pi 4 I can try but I don't think they have native OTG support?

admiralspeedy, 2024-02-17:
Alright, I did some testing.

Firstly, I realized that running my own build of the pull request was pointless because it looks like the latest artifact on here includes that code already. I ran the latest artifact build on my Pi 4 to see if the same issue (and that exception) occurred.

1.  The same connection issues apply. It does sometimes connect from a cold boot but I think the problem is related to the fact that when you open the door, the head unit powers up and sits on a ford logo until you turn the car on, so it's providing power to the Pi but it doesn't start looking for AA devices until Sync 3 actually boots. It seems like if I get in the car and immediately start it so Sync boots, it will usually connect but if I get in and wait a minute or two before starting the car, it will not connect unless I power cycle the Pi. This happens on both my Zero W and my 4B.
2. On either device, if I leave the range of the car and lose connection, when I come back to the car it will not ever re-connect, I just get "Looking for Android Auto" looping forever on my phone until I power cycle the Pi, and then it will connect.
3. I pulled the logs (below) from the Pi 4 after a failed attempt to connect and I do not see any exception like what was in the other user's Zero W logs.
4. I noticed that when it does work, the Pi 4 connects WAY faster than the Zero W. Is this because it uses 5 GHz?

Ultimately, this is basically useless for me now as I have no interest in power cycling the Pi every single time I get in my car, or if I leave my car running and go out of range. I haven't tested it but I assume if I remote start the car and don't come into range immediately, it will probably also fail to connect the same way as if I left and came back, and this is something I do often for at least 6 months of the year because I live in Canada and command start my car most days in the winter to warm it up.

I'm trying to get myself familiar enough with the code to try a few things, but what I don't quite understand is how the communication works. I see people mentioning that the car sends a request to look for an AA device, but unless I'm mistaken, is it not the other way around? Doesn't the device (phone or adapter) notify the car that it is AA capable when it's connected? It would make sense to me if it worked that way, and what I assume is happening is that the adapter in this case is only sending that report one time on boot and the head unit is missing it because it's being sent before it can actually listen for it.

I looked through the code for the reconnect pull request and I don't fully understand it, however could we just code such that whenever a phone connects to the Pi's Wi-Fi network, the Pi triggers a new connection to the head unit?

This isn't only a Ford problem, if you skim through the issues it looks like it's happening on at least Suzuki and Renault vehicles as well.

Pi 4 Log:
`Jan  1 00:00:04 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:04 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-02-11 10:29:05 UTC)
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:04 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7l (runner@fv-az979-539) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Sun Feb 11 11:02:21 UTC 2024
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
Jan  1 00:00:04 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000039bfffff]
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
Jan  1 00:00:04 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
[... cut ...]
Jan  1 00:10:35 buildroot authpriv.info dropbear[286]: Child connection from 10.0.0.12:51532
Jan  1 00:10:37 buildroot authpriv.warn dropbear[286]: Login attempt for nonexistent user
Jan  1 00:10:43 buildroot authpriv.info dropbear[286]: Exit before auth from <10.0.0.12:51532>: Exited normally
Jan  1 00:10:48 buildroot authpriv.info dropbear[287]: Child connection from 10.0.0.12:51440
Jan  1 00:10:50 buildroot authpriv.notice dropbear[287]: Password auth succeeded for 'root' from 10.0.0.12:51440`

admiralspeedy, 2024-02-18:
Tried both modes with the "select mode operation" pull request, neither works properly either. Basically every time I start the car I have to wait until Sync is booted and then power cycle the Pi.

admiralspeedy, 2024-02-20:
So, I have a bit of an update.

When I first started using this project I too went to the latest artifact but when I did that I didn't initially realize it was containing code from a pull request. I realized that late last week and decided to run a build of just the main repo this weekend (since there hasn't been a release November but there have been changes) and it appears to working completely fine for me. My phone has connected from a cold boot without issue since Saturday.

The only thing I don't understand is that the artifact mentioned in this issue should be the same code as what I built. Are you sure you downloaded that artifact and not one containing the pull request?

cr08, 2024-02-20:
> So, I have a bit of an update.
>
> When I first started using this project I too went to the latest artifact but when I did that I didn't initially realize it was containing code from a pull request. I [...]

Positive. The link I posted above is exactly where I pulled from (which was the latest at the time of the post, I see some newer artifacts are up now). I will try later this afternoon with a local build and report back.

Sorry I haven't posted any newer updates from my end. My local Microcenter went out of stock of the Zero 2's and the only other free Pi I have is a 4 which @admiralspeedy has already been working with. Will continue to try the 0w I have now with the test this afternoon.

cr08, 2024-02-20:
Ok. Just a quick test sitting in my garage and building my own image seems to have helped here. I need to take a few drives to confirm, but it's looking good so far. Voice/mic usage also seems to have been fixed in my case. Same 0w and microSD card I was using previously.

admiralspeedy, 2024-02-20:
> Positive. The link I posted above is exactly where I pulled from (which was the latest at the time of the post, I see some newer artifacts are up now). I will try later this afternoon with a local [...]

The one you linked was not the latest at the time you made this issue. [This one, created by the pull request existed before you made this post, as well as two others before the one you linked. ](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/actions/runs/7861148523) I can only assume you are mistaken on which one you downloaded because if you had truly used the one you linked, it would have been the exact same code as what you and I both built because that was the last change to the main repo until the name change this morning.

I will say I did speak too soon though. So it does work perfectly for me now every time from a cold start however I did just find a situation where it doesn't work properly now. When I got home from work I parked my car on the street for about 4-5 minutes because I had to do something in my driveway. I got back in my car and pulled it in the driveway when I was done and it was just stuck on "Looking for Android Auto..." on my phone, despite working perfectly from cold starts three other times today.

I suspect my car did not completely shut down the Pi in the 5 or so minutes the car was parked and off. For whatever reason my Escape likes to seemingly randomly and arbitrarily keep power going to the USB ports for a long time after shutting the car off, and sometimes it doesn't. When I was testing last week, it kept the Pi powered on for almost 10 minutes after I got out of the car for some reason (I knew because the AA Wireless network was still showing up on my phone in my house) and my touch screen backlight stayed lit up. Today on my lunch break at work, I shut the car off when I got back to the office and everything turned off immediately, including the Pi. 

Not sure why it does this but we definitely need some code to work around it because I suspect (haven't tested yet but I will) that this also means that if I walk out of range of the Pi with my phone and come back, it will not connect again without power cycling the Pi. I think this is what the pull request is actually trying to fix, however it seems that it breaks the initial connection as we have both experienced now.

cr08, 2024-02-21:
> The one you linked was not the latest at the time you made this issue. [This one, created by the pull request existed before you made this post, as well as two others before the one you linked. [...]

Ok. I had a chance to dig further into it and I see where the disconnect was. I was originally following the link @nisargjhaveri posted in #52 which ONLY shows main branch commits and at the time of writing THAT was current even though there was newer stuff in other branches. Ultimately I think going forward I'm just going to do my own builds and save the confusion.

The Pi not fully powering down in those short stops I can definitely see with Sync. I don't know the exact behaviors, but I do know under normal operation it shouldn't be shutting down straight away likely to allow a warm boot to happen on short trips. The age/charge level of the 12v battery can also effect this and how long it stays powered. I know when I was experimenting a while back I had an old phone connected to the second USB port on the back of the APIM for power with Tasker configured to shut down on power loss and sometimes it stayed powered for 5-10 minutes after key-off while some others it shut down shortly after. But ultimately it means this definitely seems like a key area of focus for Ford vehicles.

admiralspeedy, 2024-02-21:
Well, I can't explain why that build wouldn't work but do your own does then.

I could test it myself but it's probably not worth the effort lol. I'm trying to understand the code a bit more to figure out the power down thing.

admiralspeedy, 2024-02-23:
Well, it's like 90% functional for me. Yesterday morning I had to disconnect and connect the Pi twice on my way to work before it would connect, and now it has worked again every time since then,

We really need persistent logs because I cannot simple turn around and come back within range of my house/PC to SSH into the Pi whenever it fails and doing it on a phone is tedious.

nisargjhaveri, 2024-02-23:
> We really need persistent logs because I cannot simple turn around and come back within range of my house/PC to SSH into the Pi whenever it fails and doing it on a phone is tedious.

I completely agree. Though, frequently writing on the SD card can corrupt the SD card if the power suddenly goes away, which would be the case in most cars, we don't get a chance to shutdown properly. And once it is corrupt, you need to re-image the card to make it work again.

Maybe we can have a config for this, and when on, it'll to persist the logs. This should minimise the risk to only the people actively debugging.

admiralspeedy, 2024-02-25:
> I completely agree. Though, frequently writing on the SD card can corrupt the SD card if the power suddenly goes away, which would be the case in most cars, we don't get a chance to shutdown [...]

Hey, I see in the release notes for 0.3.0 you said "Connect to usb after phone connection is available. This reduces timeout issues in many head units." but unless I'm blind, I do not see any such changes related to this in the repo?

nisargjhaveri, 2024-03-01:
#72 should help with the issue where headunit starts much later than the power to the the dongle, if anyone wants to try out.

@admiralspeedy, @cr08 let's move the persistent log conversation to a new issue?

admiralspeedy, 2024-03-01:
> #72 should help with the issue where headunit starts much later than the power to the the dongle, if anyone wants to try out.
> 
> @admiralspeedy, @cr08 let's move the persistent log conversation to a new issue?

I will build this right now at work and try it when I leave. Do you think this mode will also resolve the issues related to the phone connecting and disconnecting while the dongle is still powered?

nisargjhaveri, 2024-03-01:
> Do you think this mode will also resolve the issues related to the phone connecting and disconnecting while the dongle is still powered?

No the reconnection issue is different. It needs some more work the cleanly close tcp and usb sockets on error or timeout. Not part of the current change.

admiralspeedy, 2024-03-01:
> > Do you think this mode will also resolve the issues related to the phone connecting and disconnecting while the dongle is still powered?
> 
> No the reconnection issue is different. It needs some more work the cleanly close tcp and usb sockets on error or timeout. Not part of the current change.

I spent most of the evening last night toying with some ideas on how to handle it, but the sockets do not throw an error or timeout when the phone is disconnected. Reading and writing from them is still successful.  I was able to set up the Android Studio Desktop Head Unit Emulator to test this in my house instead of going back and forth from my PC and my car. It's handy because I can also see logs on the Android Auto side through the head unit emulator's terminal window.

The first thing I tried is creating another thread in proxyHandler that loops and pings the phone with a system call so I could listen for it to disconnect, but this is super hacky and slow. I started looking at the emulator logs and I can see that when the phone is disconnected AA reports a timeout so I suspect that if we intercept the messages in proxyHandler we can detect those timeout messages.

I could also be really dumb, and maybe there is a really easy way to determine when the phone has disconnected, but I haven't touched sockets for years.

kman58, 2024-03-11:
> Had a chance today to go out for a few sequential trips with about 5-15 minutes in between. Knock on wood it connected reliably each time and the subsequent stops (likely the Pi was still booted/powered) it reconnected pretty quick with no input from me. Not 100% sure if it was long enough for things to fully power down at the time.
> 
> However the lack of working mic is back again and did not resolve itself on those subsequent drives. I'll check it again after it has sat and fully powered down and try again and get the logs.
> 
> Also I'll open the new issue regarding the persistent logs here soon.

Have you had any resolution with the mic input not working? I've been having the same issue on a F150 with Sync 3.4. Sometimes the first input will be recognized (ex: asking it to text someone), but then nothing else after. Most of the time though the mic does not seem to be picked up at all.

admiralspeedy, 2024-03-11:
> Had a chance today to go out for a few sequential trips with about 5-15 minutes in between. Knock on wood it connected reliably each time and the subsequent stops (likely the Pi was still [...]

That's pure luck because the current build still does not handle disconnects properly lol. I worked on it all weekend though and have made some progress. We have been discussing it on #26 .

admiralspeedy, 2024-03-11:
> > Had a chance today to go out for a few sequential trips with about 5-15 minutes in between. Knock on wood it connected reliably each time and the subsequent stops (likely the Pi was still booted/powered) it reconnected pretty quick with no input from me. Not 100% sure if it was long enough for things to fully power down at the time.
> > However the lack of working mic is back again and did not resolve itself on those subsequent drives. I'll check it again after it has sat and fully powered down and try again and get the logs.
> > Also I'll open the new issue regarding the persistent logs here soon.
> 
> Have you had any resolution with the mic input not working? I've been having the same issue on a F150 with Sync 3.4 and using a pi0w. Sometimes the first input will be recognized (ex: asking it to text someone), but then nothing else after. Most of the time though the mic does not seem to be picked up at all.

I've been talking about this issue in #26 but I found some stuff online suggesting that this issue is not exclusive to this project and happens with many of the commercially available adapters too.

Some people suggest that it is related to using 2.4 GHz over 5 GHz, but I disproved that by switching from my Pi Zero W to my Pi 4, which uses 5 GHz. I have also tried a few other code changes but had no luck.

One suggestion I did read is that someone said what fixed it on their AAWireless dongle was changing a setting in their app to keep the Bluetooth connection alive after a disconnection and enable "Use this device for calls" in the dongle's Bluetooth settings. I am going to make some changes today to keep the Bluetooth connection alive but I don't know how I'm going to do the call routing thing.

cr08, 2024-03-11:
> That's pure luck because the current build still does not handle disconnects properly lol. I worked on it all weekend though and have made some progress. We have been discussing it on #26 .

Yeah. That first day seemed to be perfect but subsequent days it went back to being hit or miss. But it did seem to be improved overall.

Also good catch on the microphone audio issue. I haven't really bothered to look much into the commercial products and how they behave. Mostly because I have a Pixel 4a with poor battery life that using wireless regularly isn't really useful. I jumped on this project to tinker and provide any help I can because the Pi was already on hand. Can't say no to free! 😅But I am hoping to get a new phone in this week so I can start using it more often.

Sorry I haven't kept up on this or provided the logs/started that persistent log issue. My day job's kept me super busy and haven't had the extra mental capacity to tinker. I have a few errands I need to run later today and I'll do some more checking and update accordingly.

nisargjhaveri, 2024-04-01:
Bunch of the changes for recovering the connection has went into v0.4.0 along with dongle mode, which should also helo with early start of dongle in some cases. I believe most of the issues should've been fixed with it.

nisargjhaveri, 2024-09-19:
Closing this for now. I believe most of the issue would be solved in the latest version. Feel free to reopen or create new issues as required.

closed 2024-09-19 by nisargjhaveri