nisargjhaveri/WirelessAndroidAutoDongle #57: Slow reaction and low resolution

MrPresident2, 2024-02-19:
Hey guys,
I have a few questions, what Pi will be best used for your project?
For now, I'm using Pi zero w2 and it seems to be extremely slow compared to the AAWireless dongle,
also it seems like the resolution is a bit lower (compared to the AAWireless dongle), is that fixable?

Great project and I can't wait to see it grow.

gamelaster, 2024-02-19:
Low resolution is not possible to change, it is resolution what should be for your screen. AAWireless is doing man in the middle proxy with SSL, where we just doing packet forwarding, so this is not possible.

MrPresident2, 2024-02-19:
> Low resolution is not possible to change, it is resolution what should be for your screen. AAWireless is doing man in the middle proxy with SSL, where we just doing packet forwarding, so this is not possible.

The answers are a bit weird, so it is not possible to make the project work faster? I should expect 5-second delay?
or is there anything that can be done to be improved? hardware or software?
also, there is a difference in resolution between AAWireless dongle and this project(not the "screen").

Ioniq3, 2024-02-21:
I tried to compiled the project in 64bits buildroot to test if I can boost the performance increase but it is really complex and after do it the udc of the usb doesn't work so I cannot be test it.

thepctech, 2024-09-17:
I have an Hyundai Tucson 2022 hybrid with the wide-screen Gen5W screen.  For some reason the wide-screen version does not support wirelessAA.  I have tried this project on a pi zero w2 to be wireless, and it's pretty impressive that it works at all.  There is a bit of screen stutter at times while driving, seems to happen less when the phone is out of my pocket, and closer to the pi.  Boot times are not bad, maybe 20 to 30 seconds.

I can confirm that the screen resolution is NOT the same, it is noticeably lower.  When you are moving you can see a noticeable difference with roads that are almost horizontal, it is also noticeable on street names near the top of the screen (smaller font).

If you claim that's not possible, I may me able to get a screenshot with it hooked up to a cable, then using this zero 2w at the same location so you can verify that this is in fact happening. 

Brian

gamelaster, 2024-09-17:
Hi @thepctech I am sorry but I don't exactly understand what you mean in the end.

Pallu08, 2024-09-29:
+1 resolution decrease and drag latency increase when using this module on a Zero 2W in a 2011 Prius.

Fantastic otherwise. No cutouts. Thx for your work

thepctech, 2024-09-29:
@gamelaster I'm not sure what you can't understand.  It has been said that this software will use same resolution that it uses if it is connected to a wire.  Like it just passes it through.  I'm saying that is not the case, there must be some negotiation with the screen resolution or a DPI setting that it uses to calculate the output resolution over wifi/BT.  

It is NOT the same resolution as when it is wired.

This is a picture taken with my phone, this is with android auto connected wireless with the Pi Zero 2 
https://www.amazon.com/photos/shared/eR9zOAwISvalauFiYomJnA.8qWqFB7MWl0Ns5fZf2AeLF

This is as close as I could get with the phone connected with a wire
https://www.amazon.com/photos/shared/rhw1OMnpSWaJZvMjSSitTw.pG1rDmp9gXW5i-JIwWXBwn

You can easily see on the roads (green and red) that the screen is not rendered the same.

Also, android auto with a wire can show two apps, wireless with the zero 2w it defaults to just 1 app, see below

Wireless
 https://www.amazon.com/photos/shared/8zYTILQ6TLKRUc4nIxAqeA.9TIYTfa5fwJs3fyJ7SEAzg

With a wire
https://www.amazon.com/photos/shared/tyeSPAM2SnunqTKccTEwGg.jZ6E024glm61Nf7rnigm31

You can see that the screen DPI/resolution with the wireless Pi Zero 2W is NOT the same.

Any questions, I can provide more info if needed.

Brian

gamelaster, 2024-09-29:
@thepctech Thanks for explanation.

This dongle acts as proxy between headunit and your phone wirelessly. Android Auto communication is all encrypted, and since this could introduce another delay and also power consumption (and some other technological things to solve), what this does is just making the bridge. Thus, this dongle is not capable of doing such changes as resolution and quality, as it sees only encrypted communication between phone and headunit. Thus, all those settings are set by phone according to information provided by headunit.

This means, that if this happens, Android Auto in app decides to decrease resolution and/or decrease video bitrate, to ensure more stable wireless connection. On cable, this is not case, as speeds of transports are mostly stable on wired connection. 

Same thing is about default layout of system, all decided by phone (Android Auto app).

thepctech, 2024-09-29:
So I'm not clear on what you are say.  It's clearly a lower resolution over
wireless on the PI Zero 2W.  Are you saying that I need to use a faster PI,
maybe a PI4, that has better wireless and Bluetooth speeds?

It's clearly rendering the screen at a different DPI.  Maybe android auto
is dropping the resolution to keep the frame rate up??

gamelaster, 2024-09-29:
@thepctech we do not know, as it is all controlled by Android Auto app. Maybe more powerful RPi or using 5 GHz (But we do that already as far as I remember) might help. But I doubt it will help, it is configured like that to be stable on most devices and Headunits.

thepctech, 2024-10-06:
I'm trying it with a Pi 3A to see if the extra .5 GHz in processor speed will make a diff.  Not sure if there is better wifi/BT on the 3A, but if so maybe that will help too.

thepctech, 2024-10-06:
So....

I set up an 3A+ and it connects on the 5Ghz Wi-fi standard zs the 2.4GHz that the Pi Zero 2W has.  And I have what looks like full screen resolution, the same as the wired equivalent.  So I would agree 100% that Android Auto on the phone and Android Auto on the head unit negotiate some speed during the initial connection.  The 2.4GHz might be enough for a smaller auto head unit, but the widescreen needed for my auto requires a bit more than 2.4GHz can provide.

Note:  It's not so easy  to find a USB-A to USB-A cable needed for the Pi A+

gamelaster, 2024-10-06:
@thepctech Nice! I am sorry, for some reason I thought that Pi Zero 2W had 5 GHz WiFi as well, so I did not count with this...
By specification from 2018, Android Auto supports only 5 GHz WiFi (officially), due it less interrupts other WiFi's and have more bandwidth. It's very likely that if Android Auto sees that it's connected to headunit through 5 GHz WiFi, it will allow higher bitrate on the video.

nisargjhaveri, 2024-11-05:
Thanks @thepctech for sharing the observations! This helps to know Android Auto might be differentiating based on 5GHz wifi.

manio, 2025-03-31:
FWIW, if you're interested, thanks to @gamelaster help, the DPI change is now possible via `--dpi` parameter in `aa-proxy-rs` (at least it's working in my Renault HU):
https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#dpi-settings