nisargjhaveri/WirelessAndroidAutoDongle #58: Pioneer head unit.

guarddog13, 2024-02-21:
It works with the head unit. However as an fyi you must update the date via ssh in the terminal on the dongle, not just the country code. 

On a side note..thank you for this project. It serves 2 purposes for me. 

1. My wireless head unit has been having connectivity issues and I didnt want to buy a new one or lose AAWireless. Raspberry Pi 4b laying around and a 3d printed case. The otg port powers the dongle as well as provides AA to my head unit. I have a system in my car and will say the sound is superior than the HUs wireless. The pi must be doing the processes vs the phone? Or the radio vs the phone? Idk but using my tuner I can see the actual difference in sounds coming from my speakers at similar levels. 

2. I drive company cars that often don't have AAWireless, which is tough when you're in and out of a car all day. Now i have a portable dongle to carry with me in each car and wont have to even pair it to each car. 

Again thank you!!!

nisargjhaveri, 2024-02-21:
Thanks for the note! No idea why the sound would be different though without more details.

Is there a reason why you had to update the date as well? The date gets reset on reboot and shouldn't actually matter.

guarddog13, 2024-02-21:
I'm not sure. When I kept trying to connect (only the first time) the head unit would make a comment to update the phone's date to match the HU date. Once connected the message quit.

I'm assuming either my connection was that bad prior or the HU processes USB vs a wireless connection differently.

nisargjhaveri, 2024-02-29:
Right, could be some difference somewhere, not really sure :)

Even if you set the date via ssh, it is not really persisted and would reset the next time Pi boots. Most likely the date mismatch was between the head unit and the phone and has nothing to with the dongle.

I'll close this issue for now. Feel free to reopen. Glad that the project was helpful! :)

closed 2024-02-29 by nisargjhaveri