nisargjhaveri/WirelessAndroidAutoDongle #59: Rebooting when android auto on car attempts to connect Skoda Kodiaq 2018

gazd25, 2024-02-23:
Hi,

Thanks so much for all your hard work on this project, I'm trying to use your ready made sd card images to work with my Skodi Kodiaq 2018 in the UK.

I've tried with both a Raspberry Pi 0 w v1.1 and also a Pi 0 W 2 (using the correct images 0.2.0 and after editing country code) and both with pretty similar results in that when I connect to the car usb port it sees the dongle initially but doesn't auto connect, so after a little time I hit the button to tell it to connect which it initally seems to do, but then shortly afterwards the rpi seems to reboot because I see the green light flashing.

I have connected using the wireless and downloaded the logs from messages right before the reboot and I've pasted in a file of what I got on my last time.

Here is what I've tried so far:

Both different types of hardware = same
Different cables = same
Rolled back to v 0.1.1 = very slightly different in that on a couple of occasions I managed to see the Bluetooth from the rpi, but it soon stopped once the reboot occured.

The error I see on my car is smartlink not supported or similar.

Is there anything you can think of for me to try to get this working correctly using either of my devices? I'm happy to troubleshoot with you where I'm able.

Thanks again for your help.
[rpi 2 w logs.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14386677/rpi.2.w.logs.txt)

MrCrrispy, 2024-02-23:
Probably a timeout issue like most other cars, try using the latest automatically generated images from [here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/actions/runs/7978244678)
Scroll down to the Artifacts section and download the zip.

gazd25, 2024-02-23:
Thanks for the response, I'll give this a try over the weekend when I have some free time and feedback to you how I get on.

gazd25, 2024-02-24:
The version from the zip worked nicely right away using the same process.

I downloaded the 0.2.0 version because the link presented on the readme shows 0.2.0 as newest and I didn't realise there was newer ones to use.

Thanks again for the help getting this working and creating such a great open source tool, very much appreciated.

nisargjhaveri, 2024-02-25:
Glad that it worked with the latest builds. I've also released 0.3.0 which should also work for you.

closed 2024-02-25 by nisargjhaveri