nisargjhaveri/WirelessAndroidAutoDongle #61: RPi02w v0.3 prebuilt wifi not activating

cjhnz, 2024-02-24:
Tried the latest build v0.3 and I now get the option to pair on the bluetooth but the wifi does not show up.  Confirming going back to v0.2 I can't pair the bluetooth but can connect to the wifi.

nisargjhaveri, 2024-02-25:
Would it be possible to get the logs from Rpi?

cjhnz, 2024-02-25:
Not sure how I can get the logs off without access to the wifi.  I found that when I plugged it into a PC this version tells me that the USB device is not recognized but previous versions it was reconized.

nisargjhaveri, 2024-02-29:
Interesting. Missed that part! :)

But usb also not working seems suspicious as well. In this case you might need to build locally and change location of the logs. Or connect a screen and see what's going on.

cjhnz, 2024-02-29:
Hooked it up to a screen to watch the bootup and found this error

2.041087] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430b0-sdio for chip  BCM43430/2  
2.052112] usbcore: registered new interface driver brcmfmac  
2.054961l brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430b0-sdio.ra  spberrypi ,model-zero-2-w.bin failed with error -2 
2.060727] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430b0-sdio.bin failed with error -2

Cheers

nisargjhaveri, 2024-02-29:
Looks like we're missing the proper firmware files post kernel upgrade.

nisargjhaveri, 2024-02-29:
You can try the build from #69 when available, that should add the required firmware files we're missing.

cjhnz, 2024-03-01:
Awesome work!  Build #69 is working great.  Connects and stays solid.  No lag with spotify or anything that I have noticed so far.  I have had a couple of times it has not connected properly.  I think this maybe due to the headunit not booting enough for it to activate AA.  Will see of I can find a pattern to it

nisargjhaveri, 2024-03-01:
Fixed with #69.

closed 2024-03-01 by nisargjhaveri