nisargjhaveri/WirelessAndroidAutoDongle #62: Not working in MB Class A W177 (MBUX)

siserte, 2024-02-25:
My car is not showing the Android Auto App when I connect the RPi.

I attach the log:

[messages.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14395875/messages.txt)

nisargjhaveri, 2024-02-25:
Seems like the phone was not connected over bluetooth. Once the Pi boots up, you need to pair and connect the bluetooth from phone. Is that giving some errors?

siserte, 2024-02-25:
I expected that my car showed something regarding the documentation:

"Once the car recognizes the device and sends a request to start Android Auto, bluetooth is enabled."

I cannot connect to the Bluetooth since there is no device available.

Thanks.

nisargjhaveri, 2024-02-29:
"Once the car recognizes the device and sends a request to start Android Auto, bluetooth is enabled." This has changed in recent versions. We enable bluetooth first now.

nisargjhaveri, 2024-04-01:
Please try out the new versions available now, a bunch of changes have went in, might help in this case depending on what the original issue was. In any case, closing the issue for now as I think the newer versions should fix this. Feel free to reopen if you're still facing the issue.

closed 2024-04-01 by nisargjhaveri