nisargjhaveri/WirelessAndroidAutoDongle #63: PI Zero W + v0.3 + Skoda Karoq (2021)

kipperfish, 2024-02-26:
Hi,

I just got a PI Zero W and installed v0.3. 
I paired my phone via bluetooth and the "AA Wireless Dongle" shows on my cars dash but my phone says 
"Connected to Android Auto" then shortly after it sits at "Looking for Android Auto"

I changed hostapd.conf from IN to GB as I'm in the UK, and the logs show the following

Any ideas on what I'm doing wrong?
 
[messages.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14407441/messages.txt)

nisargjhaveri, 2024-02-29:
Seems like car is not responding once we enable the USB after phone is connected. Could you try using v0.2.0 once?

kipperfish, 2024-02-29:
Hi, thanks for getting back (and for all your hard work!)... I just tried 0.2 and unforuntely the PI doesn't show up on the car nor is it discoverable as a bluetooth device. I've attached the logs. Please let me if I can help further, or try anything.

PS: I haven't changed the wifi from IN to GB with 0.2, but wifi seems fine (that's how I got the below logs)

[messages_20240229.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14454048/messages_20240229.txt)

nisargjhaveri, 2024-04-01:
One of the common reason for car not responding on usb has been faulty usb cables. Please make sure the USB cable is a data cable and works well?

If not the cable, need to investigate more.