nisargjhaveri/WirelessAndroidAutoDongle #64: Issue connecting phone to Rasberry pie W0

chanakyaraichur, 2024-02-26:
Hi,

 I have tried stable build 0.20 and latest build on Rasberry pie W0 but it's not working .
Android phone shows AAWireless in list but after connecting to it nothing happens . 
Even car head unit as well don't display info when connected to android auto port .

I am using Hyundai venue 2023 model . 

Have you came through this issue before ?
Is there any way to over come this problem ?

Can you provide video with steps so that we can also follow the same steps or check what mistake we are doing ?

Thanks and Regards,
Chanakya

milindsoni, 2024-02-26:
Experiencing a similar problem with the Hyundai Venue 2023 model. When attempting to use wireless Android Auto, it successfully shows the Bluetooth connection, but after establishing the connection, nothing happens.

nisargjhaveri, 2024-02-27:
This can sometimes happen if the car does not actually start AA in time even after usb is enabled from the dongle. If possible, please include the logs to know more.

nisargjhaveri, 2024-04-01:
You can also try out the v0.4.0 which has some changes which might help in this case.

nisargjhaveri, 2024-09-19:
We've been seeing a common issue with bad usb cables. I believe using a good data cable would solve this issue. Closing for now, feel free to reopen as needed.

closed 2024-09-19 by nisargjhaveri