nisargjhaveri/WirelessAndroidAutoDongle #67: Raspberry Pi Zero 2W - Bluethooth paris but then disconnects

tracethepackets, 2024-02-29:
I have a brand new RPi Zero 2W and I used the pre-built image for that board but I am not able to keep a steady bluetooth connection to the deive.  Here are the steps I perform.

- Downloaded the pre-built v0.3.0 raspberrypizero2w-sdcard.img.xz image
- Copied image to micro SD card using Rufus
- I plug in USB cable to USB OTG port, it powers on just fine. 
- I see the "A A Wireless Dongle" bluetooth device and I am able to successfully pair to it.  
- Within less that 5 seconds I loose that bluetooth connection to the RPi zero 2W
- If I try to reconnect, it fails. I have to unpair and start over, but I just end up with the same end result.

I have attached a copy of my messages file contained with /var/log folder.  

Anyone able understand what went wrong? 

[messages.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14441344/messages.txt)

nisargjhaveri, 2024-02-29:
We turn off the bluetooth once the AA connection sequence via bluetooth is complete. After this, we enable usb and wait for the car to connect. Seems like that is not happening from the logs.

nisargjhaveri, 2024-02-29:
You can try once with v0.2.0 in case it helps.

tracethepackets, 2024-02-29:
> You can try once with v0.2.0 in case it helps.

Thank you. I will give v0.2.0 a try to see if that works any better.  With respect to your previous response above re: turning off Bluetooth once bluetooth is complete, it would appear that the RPi unit is working as expected but there is some sort of compatibility issue with my headunit not detecting the device?

nisargjhaveri, 2024-02-29:
> it would appear that the RPi unit is working as expected but there is some sort of compatibility issue with my headunit not detecting the device?

Yes, something like that.

> is it possible that a faulty USB cable is causing the RPi to not be detected correctly by head unit?

Yes, this is very much possible. We'd try to connect to phone simply on boot regardless if the USB can be connected or not. If the cable is faulty or power-only cable, we'll see the same behaviour.

thecorradokid, 2024-03-01:
I have similar symptoms on my Zero 2W running pre-built image 0.3.0. I can pair with Bluetooth the first time, my phone screen flashes as it initializes Android auto, and my head unit begins to connect. The Bluetooth connection is killed before Android Auto shows on my head unit and the head unit stops recognizing the device. Once the head unit stops recognizing the device, I can reconnect with Bluetooth and start the process over again with same results.

I did try 0.2.0 and had better results. I was able to initialize Android Auto, however if I power cycle the car and Raspberry Pi, I cannot get it to connect automatically again.

2019 Volkswagen GTI with MIB II
Pixel 7 Pro

nisargjhaveri, 2024-03-01:
The first failure, usb seems to take more than 10 seconds to connect and tcp timeouts by then. The second failure, the usb never connected, at least until the log ends.

If possible, try out the builds from #72 and see if that helps?

mariolugo2, 2024-03-01:
I have the same problem, I am testing with version 0.2.0 and see what results it gives me

RPi Zero 2W
Volkswagen GTI 2017 con MIB II
Pixel 7 Pro

> Tengo síntomas similares en mi Zero 2W que ejecuta la imagen prediseñada 0.3.0. Puedo emparejarme con Bluetooth la primera vez, la pantalla de mi teléfono parpadea mientras inicializa Android Auto y mi unidad principal comienza a conectarse. La conexión Bluetooth se interrumpe rápidamente antes de que aparezca Android Auto en mi unidad principal y la unidad principal deje de reconocer el dispositivo. Una vez que la unidad principal deja de reconocer el dispositivo, puedo volver a conectarme con Bluetooth y comenzar el proceso nuevamente con los mismos resultados.
> 
> Probé 0.2.0 y obtuve mejores resultados. Pude inicializar Android Auto, sin embargo, si apago y apago el auto y la Raspberry Pi, no puedo lograr que se vuelva a conectar automáticamente. Por lo tanto, se confirma que el cable funciona.
> 
> Volkswagen GTI 2019 con MIB II Pixel 7 Pro
> 
> EDITAR: descubrí cómo tirar de los registros. Reinstalé 0.3.0 y demostró el comportamiento anterior (archivo "fail1"). Hice un ciclo de energía y esta vez se conectó y funcionó correctamente (archivo "éxito"). Esto continuó durante otro ciclo de encendido hasta que finalmente comenzó a fallar nuevamente (archivo "fail2").
> 
> [mensajes_0.3.0_290224_fail1.txt ](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14456275/messages_0.3.0_290224_fail1.txt) [mensajes_0.3.0_290224_fail2.txt ](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14456276/messages_0.3.0_290224_fail2.txt) [mensajes_0.3.0_290224_success.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14456277/messages_0.3.0_290224_success.txt)

tengo el mismo problema 

Volkswagen GTI 2017 con MIB II
Pixel 7 Pro

mariolugo2, 2024-03-02:
mib detects android auto, but does not connect to the bt of the rpi

> I have the same problem, I am testing with version 0.2.0 and see what results it gives me
> 
> RPi Zero 2W

[v0.2.0.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14467549/v0.2.0.txt)

mariolugo2, 2024-03-02:
> mib detects android auto, but does not connect to the bt of the rpi
> 
> > I have the same problem, I am testing with version 0.2.0 and see what results it gives me
> > RPi Zero 2W
> 
> [v0.2.0.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14467549/v0.2.0.txt)

Curioso realice cambios a la nombre a la red wifi y bt como menciona en este post

https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/42#issuecomment-1913269573

Genere una nueva imagen con la última rama y ya funciona correctamente hasta el momento.

Muy buen aporte.

tracethepackets, 2024-03-03:
> You can try once with v0.2.0 in case it helps.

Hi @nisargjhaveri - I gave v0.2.0 a try but that actually didn't work for me at all.  I wasn't even able to get the AAWireless Bluetooth going but I could see the Pi led flashing to I know it was booting up.  I decided to give the lateste .0.3.0 a try again but this time go through several different USB cable.  I finally hit the jackpot and now it seems to be working just fine.

There was one day where it did not activate Android Auto but after shutting off the vehicle (Honday Odyssey 2019) and turning it back in, the connection was established.  So far so good.

Thank you very much for bringing this to us.

simquad, 2024-03-19:
I've been exploring this myself on my Kia eNiro and am seeing the same behaviour.

I'm using a PiZeroW, running 0.30.1

I will share the logs when I'm home at my laptop.

Are there things I can investigate and debug?

I took a look at Bluetoothctl and if I try to connect to my phone via BT on the device registered, it suggests that it is powered down and doesn't work; is this because you're doing something to turn Bluetooth off?

Happy to help if I can

simquad, 2024-03-19:
[001.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14655550/001.txt)
[002.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14655551/002.txt)
[003.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14655552/003.txt)
I've added some logs which I think demonstrates the same issue you've been discussing here.

simquad, 2024-03-20:
> The logs suggests that the bluetooth connection was successful and it successfully connected via Wifi as well. But once the USB connection didn't succeed. The car never sent accessory start request, or something else went wrong.
> 
> The bluetooth gets disconnected once the Wifi connection is successful. So, bluetooth disconnecting is not an issue here.

Okay, that's good to know then.

I've not explored into this before, do you have any pointers on things to check, or indeed can you summarise your workflow for testing this out?

More than happy to put some time into investigating what's going on, but any 101 you have would be handy.

I'll also take a look at the Android docs.

Thanks for getting back to me.

tracethepackets, 2024-03-21:
@simquad - You may want to try different USB cables. Make sure they are rated as "data" cables and not only for charging.  I had this same thing happen to me and once I went through several cables, I found one that allowed my vehicle to make a successful connection to the Pi.  With that said, every-so-often, it doesn't make the connection on the first attempt. It may take several attempts, but for the most part, it has been working for me.

algoz098, 2024-03-22:
I used the ones I use to program IOT chips like ESP32, ESP8266 and arduinos, so that's prob not the problem. Regardless i will try again

nisargjhaveri, 2024-04-01:
v0.4.0 has several changes might help in some of these cases. Other common issues include faulty usb cable or wrong usb port on the RPi as @tracethepackets mentioned.

In any case, I feel we've been talking about multiple potentially different problems in this issue. I'll close this issue, feel free to open new issues if the problem still persists.

closed 2024-04-01 by nisargjhaveri