nisargjhaveri/WirelessAndroidAutoDongle #71: Unable To Connect to Mazda 3 2015 via Pi Zero 2 w

NabeelUppel, 2024-03-01:
Hello! 

I'm having issues.
I manage to connect to the bluetooth initally, my phone flashes for a second and I get a AA notification saying searching. My phone manages to connect to the wifi AAWirelessDongle (via Android Auto) [That's what is says under the wifi name.]

I managed to get the logs but tbh I'm not sure what i'm looking for. 

Any advice? Thanks in advance

Car: Mazda 3 2015
Pi: Pi Zero 2w
Release: v0.3.0
Phone Samsung S22+ 

[AAWirelessDongleLogs.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14463947/AAWirelessDongleLogs.txt)

nisargjhaveri, 2024-03-01:
You can try the build and instruction from #72 in case it helps.

NabeelUppel, 2024-03-02:
That did not work unfortunately. 

One thing I noticed, for car with a wire if I connect for the the first time to android auto I'll have to disconnect and reconnect for it to work. But everytime after that it connects fine. 

I wonder if that has something to do with it so I connected initially like your instructions said but it didn't work. (Same issue as before) And then I restarted the Pi but no luck

I can't say I noticed it before but now the AAWirelessDongle (connected via Android Auto) wifi keeps disconnecting and reconnecting. 

I'll add some logs. 
[InitialLogs.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14467937/InitialLogs.txt)
[RestartPiLogs.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14467938/RestartPiLogs.txt)

NabeelUppel, 2024-03-03:
It seems like Mazda's infotainment system requires that's the phone be paired to Bluetooth and USB connection for it to work. 

Could this be causing the issue? 
Is it possible to connect the dongle to the car's Bluetooth on startup? 

My car also the ability to WiFi. I wonder if it's something that can be used to connect.

nisargjhaveri, 2024-04-01:
Both the log files seems to be partial. Would it be possible to share full logs?

You can also try v0.4.0, in case it helps. With it, the phone needs to be connected via usb first, and then the bluetooth needs to be connected to the head unit as well to start the wireless connection via the dongle.

NabeelUppel, 2024-04-01:
Here's my latest logs running v0.4.0.

The phone now flashes every +-14 secs and then disconnects.
Which is more than I can say for the old version which would just fail once and give up. So progess I guess lol.

One thing to note about my car, when I connect my phone via USB to Android Auto it automatically connects to the bluetooth. I don't if that is relevent but might be worth noting. 

[AALogs_01-04-24.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14828998/AALogs_01-04-24.txt)

NabeelUppel, 2024-04-03:
My phone is already connected to my car. Should I connect my car to the raspberry pi? 

So this is the flow.
1. Connect my phone to android auto via cable -> phone connects to  car bluetooth automatically. 
2. Disconnect my phone and plug in Pi
3. Connect my phone to Pi via Bluetooth.
4. Should work but it doesn't.

nisargjhaveri, 2024-04-03:
The logs suggest that the phone connects to the RPi correctly. And RPi enables the USB. But the car is not detecting that and sending the accessory start request.

This issue seems to be happening in multiple cars, not sure with the same reason or different reasons. But I would require some help debugging this, as I cannot repro this issue. Some parts of #34 might be relevant, not sure.

NabeelUppel, 2024-04-03:
I suspect with my car, it has to be connected to device via Bluetooth. 
Is connecting the RPi directly to the car possible? 

So the setup would be.
Phone -> RPi -> Car

Currently I think it's like 
RPi <- Phone -> Car

nisargjhaveri, 2024-04-04:
> I suspect with my car, it has to be connected to device via Bluetooth.
Is connecting the RPi directly to the car possible?

Connecting RPi to car via Bluetooth is not be required, and won't help. The phone needs to be connected to both RPi and Car and that is expected.

NabeelUppel, 2024-04-04:
Good to know.  
Could it be issue with my cable? I'm not 100% it's a data cable

nisargjhaveri, 2024-04-04:
Yes. Please make sure:
1. The cable is good quailty data cable and not power-only cable
2. You're using the OTG enabled usb port on the RPi, and the power only port.

closed 2024-04-04 by nisargjhaveri