nisargjhaveri/WirelessAndroidAutoDongle #74: 2021 Kia Sorento, Raspberry Pi 3B+, One Plus 7T stuck at searching for Android Auto

sureiguess, 2024-03-03:
The title pretty much describes it. I can connect bluetooth between my phone and the pi but that is where it stops. The image is where it stops. I have attached the logs, any help would be appreciated! Thanks
[Kia_hostapd_log.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14471799/Kia_hostapd_log.txt)
[Kia_message_Log.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14471801/Kia_message_Log.txt)

[image]

nisargjhaveri, 2024-03-03:
Raspberry Pi 3B+ is not supported, it does not have USB OTG capability.

sureiguess, 2024-03-03:
My bad, thank you. I will try a different model.

closed 2024-03-03 by nisargjhaveri