nisargjhaveri/WirelessAndroidAutoDongle #77: Not connecting to ioniq 5

blondini23, 2024-03-08:
Pi0 seems to boot and offers a wifi address but then nothing happens, I would ssh into the pi but can't get in using juicessh to root@10.0.0.7 is this the correct way to get in?

admiralspeedy, 2024-03-08:
The address to SSH into the Pi is 10.0.0.1.

blondini23, 2024-03-08:
Thanks that let's me in so I can now look at tge logs

nickbruton, 2024-03-11:
Am I allowed to  add tge contents of /var/log/messages?

nisargjhaveri, 2024-03-11:
> Am I allowed to add tge contents of /var/log/messages?

Yes, please share the logs here if you're also facing similar issues.

nickbruton, 2024-03-11:
/var/log/messages

Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-03-01 07:28:45 UTC)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61 (runner@fv-az1269-741) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 Fri Mar  1 08:00:43 UTC 2024
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:5F:26:E9 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 340196K/421888K available (7906K kernel code, 702K rwdata, 2184K rodata, 256K init, 498K bss, 16156K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000006] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000055] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000158] bcm2835: system timer (irq = 27)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000946] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001443] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001523] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060290] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060396] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060584] LSM: Security Framework initializing
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060797] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060867] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.062264] cgroup: Disabling memory control group subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064143] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064225] cblist_init_generic: Setting shift to 0 and lim to 1.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064550] Setting up static identity map for 0x8220 - 0x8258
Jan  1 00:00:03 buildroot kern.info kernel: [    0.065801] devtmpfs: initialized
Jan  1 00:00:03 buildroot kern.info kernel: [    0.076658] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
Jan  1 00:00:03 buildroot kern.info kernel: [    0.077054] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... cut ...]
Jan  1 00:01:26 buildroot authpriv.info dropbear[222]: Exit before auth from <10.0.0.7:58994>: (user 'root', 0 fails): Exited normally
Jan  1 00:01:28 buildroot kern.info kernel: [   88.128143] read descriptors
Jan  1 00:01:28 buildroot kern.info kernel: [   88.130156] read strings
Jan  1 00:01:29 buildroot authpriv.info dropbear[223]: Child connection from 10.0.0.7:49748
Jan  1 00:01:29 buildroot authpriv.notice dropbear[223]: Password auth succeeded for 'root' from 10.0.0.7:49748
#
nick@pi5:~ $

nickbruton, 2024-03-12:
I can't see any errors that might indicate why it doesn't work

nickbruton, 2024-03-13:
I didn't realise it needed to connect with WiFi and Bluetooth but suspect thus will make it work

nisargjhaveri, 2024-03-13:
You don't really need to connect to Wifi manually, but bluetooth needs to be paired the first time.

nisargjhaveri, 2024-04-01:
Looks like this might be fixed now. Closing the issue for now, feel free to reopen if this is still an issue.

closed 2024-04-01 by nisargjhaveri