nisargjhaveri/WirelessAndroidAutoDongle #78: Not connecting to MG4

GeralddenEngelsman, 2024-03-09:
Hey All,

I seem to get stuck with connecting to my MG4.
I tried using:
- Raspberry pi zero w 2
- Raspberry pi 4 2GB
- Raspberry pi 4 4GB

The results varied, from being able to connect to bluetooth (only on the pi zero w image 0.3) to the raspberry pi 4 saying it's not compatible with what's on the sd card (same sd card doesnt give that problem on the 2gb).

However, the pi 4 units all do not even broadcast the bluetooth, and the zero w disconnects instantly when connected on bluetooth.

I want to debug, and did setup a manual connection to the known SSID. but I am not finding that network.

So a few questions:
1. Is the ssid supposed to be broadcasted?
2. Did anyone get this to work on an MG4
3. Connecting to the 4GB raspberry pi 4 made it blink yellow 4x slow, and 4x fast, which suggests incompatible operating system, is that supposed to happen?
4. I tried flashing the sd cards with raspberry pi imager, but also win32 diskimager, results are the same. Are those correct ones to flash the sd cards with?

Gdenge, 2024-03-11:
Can you elaborate on the sequence of events that will happen on the raspberry pi

I tried using 0.31, 0.3 and 0.2 all have the same outcome when I use a raspberry pi 4 (2gb).
Am I supposed to see the BT AA_Wireless first, or the SSID, or are both visible at the same time? 

Same question for the Zero 2W.

nisargjhaveri, 2024-03-11:
In versions 0.3.0 and 0.3.1, you should see both the Wifi and Bluetooth at once. But you don't need to connect to the Wifi manually. Just pair the bluetooth and it should work. The bluetooth gets disconnected just after connecting and that is okay.

In 0.2.0, the bluetooth will be visible once the headunit starts the usb connection first. Again, you just need to pair the bluetooth for the first time.

If it is not working, you can get the logs out by connecting to the Wifi directly and using ssh to get the logs.

Gdenge, 2024-03-12:
Well, for now the issue seems to have been resolved.
I flashed V0.3.1 onto a raspberry pi zero 2 W using the win32diskimager as I had done before.
Cleared out all mobile phones on the bluetooth connections (previously only removed the one I was using and left my wife's phone in there).
After a moment, it got available on BT, then connected it and it worked.

So far only a few reboots have been done, but have been succesful. 
I will keep this ticket open for now, will come back in a day or two with updates :)

Gdenge, 2024-03-14:
Update from my end.

It keeps working, however I do often need to "reset" after initial bootup.
I think my car powers on too slow.

When I unlock my car, the USB power goes on (It's fully electric).
But only when I open up my door, my infotainment system will power on.

When "starting" the car, which is nothing more than setting it in drive mode, it will not reset the USB power like some other cars.

Would it be possible to setup like a file to "wait" with continuing to boot as others might have the same problem and boot times can be different from car to car, or to loop when it's not succesfully connected to the car with a configurable amount and wait time per loop?

nisargjhaveri, 2024-03-15:
Yes, we can add retries. But #72 might also be useful in this case. With this, the phone would wait for the headunit's bluetooth connection before trying to connect to dongle via wifi.

Gdenge, 2024-03-18:
Although I like the idea, and in some cases this could result in the desired outcome, my car is already making a BT connection when I leave my house. It's BLE will connect even if I haven't unlocked the car yet.

closed 2024-03-22 by Gdenge