nisargjhaveri/WirelessAndroidAutoDongle #82: Problem to connection with MB A180

JVerPi, 2024-03-15:
Hello community,
I have some problem to work with my mercedes A180. I have tried with diferent PIs (Pi Zero W/ Pi Zero 2W) with the same result. I can connect it by Bluetooth to 'AA Bluetooth Dongle' but Android Auto don't start. In my phone is notificacion ' Searching Android Auto' and this stay always...
There is some things to do or I make something false ?

Ioniq3, 2024-03-16:
You hace to follow the instructions .

Connection to dongle mode requires the following steps:

Clear "Previously connected cars" from Android Auto. Clearing app cache might also be required.

"Forget" old bluetooth dongle from phone.

Connect phone to headunit using a wire. Make sure wired Android Auto starts.

Remove the cable from phone, plug in the dongle.

Pair with the "AndroidAuto-Dongle" via bluetooth.

JVerPi, 2024-03-18:
Hello ioniq3,

Thanks you very much for the instrucctions. That's work fine. 

Only I should be deleted/forget  the previous Bluetooth of the car and after make a restart in the phone. Then try again with the AA dongle and worked fine. 

Thanks again for the help.

closed 2024-03-18 by JVerPi