nisargjhaveri/WirelessAndroidAutoDongle #83: Problem with Kenwood head unit detecting RaspberryPi4

Paul680qwe, 2024-03-17:
Hello, I installed the img onto the SD Card using Raspberry Pi Imager. I connect it to my car and it powers on. I can connect my Pixel 7 to the Pi, but the Kenwood head unit installed in my Dodge Caravan 2017 wont detect the Pi. I don't have a laptop to connect to the pi while its in the car so I can't ssh in. Any suggestions on a fix?

Paul680qwe, 2024-03-20:
I know that it doesn't work as of right now, but would it also be possible to use the wifi connection as a hotspot taking in ethernet? If so how?

nisargjhaveri, 2024-03-20:
> I know that it doesn't work as of right now, but would it also be possible to use the wifi connection as a hotspot taking in ethernet? If so how?

Could you explain some more on what you're trying to achieve?

Paul680qwe, 2024-03-20:
I want to have the raspberry pi import data via USB tethering and distribute it to the devices on the WiFi Connection.

nisargjhaveri, 2024-04-01:
> I want to have the raspberry pi import data via USB tethering and distribute it to the devices on the WiFi Connection.

This might be possible and you should be able to find more info online. Is this related to this project though?

For the original issue, is it still an issue with v0.4.0? If so, would it be possible to share the logs?

Paul680qwe, 2024-04-21:
Sorry for the delay, I used 4.0 and it worked. Now I want to try an make a hotspot using a phone with an ESim. Any idea on how I would be able to do that?

nisargjhaveri, 2024-04-21:
That's great. Thanks for updating.

You can have a look at #46 on some pointers on how to connect to external Wifi instead of the hotspot from the RPi.

closed 2024-04-21 by nisargjhaveri