nisargjhaveri/WirelessAndroidAutoDongle #85: Assistant Voice Command Issues

admiralspeedy, 2024-03-18:
Based on my testing and research, I'm pretty sure the issue I'm having with Assistant (not hearing me sometimes at all, but mostly it just hears 30-50% of what I said) is not related to this project specifically. Sometimes it works fine, sometimes it doesn't and I'm trying to figure out why.

I believe it is a bug caused by some combination of car, phone, Google app version, and Android Auto app version but I thought it might be helpful for all of us to compare setups and see who is and isn't having issues.

My current setup is the following:

- 2019 Ford Escape SEL, Sync 3.4
- Pixel 7, Android 14 (AP1A.240305.019.A1, March 5th Security Update)
- Google App 15.10.44.29.arm64
- Android Auto 11.4.640814-releawe

So far I have tried the Google app beta, I have downgraded Android Auto to 10.0.X and 10.6.X, I have cleared data and cache on both the Google and Android Auto apps, and I have rebooted my phone when the issue is happening. I have also tried using my Pi 4 instead of my Zero W to take advantage of 5 GHz Wi-Fi and I unrestricted battery usage on both the Google and Android Auto apps.

None of these specifically seems to fix the issue, but there have been times where Assistant just works perfectly and then suddenly it stops again. 95% of the time it does not work properly and I can't sent more than a 3-5 word text because it either hears me wrong or just doesn't hear most of what I said at all. It works flawlessly when using Android Auto wired.

Doing research, I have noticed that tons of people have this issue with all different wireless adapters, including the MA1 and AAWireless, and I have found a couple comments from people saying they experience the issue with native AA Wireless in their car not even using a dongle.

One thing of note that I did read about this issue is that some people noticed that if they went back and checked the recordings that Assistant saved, that the recording sounded very sped up (like a weird bitrate issue) and only when using AA in wireless mode, however I cannot verify that this is what is causing the issue because even though I have recording enabled, Assistant does not seem to save recordings of anything I say to it in my car for some reason. It saves one every time I use it on one of my Google Home speakers, but never in my car so I can't check.

Ioniq3, 2024-03-18:
Yes, I have the same problem, in a commercial carlinkit wireless dongle is the same. I read that the assistant voice command use the microphone of the phone instead of the bluetooth connection of the car, so depends on the physical situacion of the phone, capture better the sound or not, but I don't find official description about it.

admiralspeedy, 2024-03-18:
> Yes, I have the same problem, in a commercial carlinkit wireless dongle is the same. I read that the assistant voice command use the microphone of the phone instead of the bluetooth connection of [...]

It's not using the phone mic, I actually already confirmed this by talking directly into my phone both normally and loudly, thinking this may be the issue (since otherwise it would be in my pocket where the mic would be obscured) and it made no difference.

I actually just went through my Assistant recordings again and found some that finally did record when I used Assistant while driving and they all sound totally fine, not sped up or broken and all clearly from my car's microphone so I'm really stumped as to why it is happening at all.

nisargjhaveri, 2024-04-01:
If this is happening with other dongles as well, it might be an issue with Android Auto itself and not this project. Is this still an issue?

admiralspeedy, 2024-05-08:
> If this is happening with other dongles as well, it might be an issue with Android Auto itself and not this project. Is this still an issue?

It has actually gotten worse for me recently. About 80% of the time now Assistant will now just not work at all until I disconnect and reconnect or switched to wired. It just acts like I'm not speaking, but phone calls work totally fine.

dvelabs, 2024-05-09:
> I am also facing issues with voice commands. Seems that it can not hear me for some strange reason I am on V4 and the problem is common across assistant itself, google maps search and spotify. I had recording saving disabled on my google account so i can't check what/if recording was made. Just enabled it again in order to see if the recording is made and sent to google. Will report back

Just checked "My Activity" on google account page.
Seems that even though assistant is triggered, recording does not reach google.
[image]

When using my assistant on my phone, the recording is stored.
[image]

admiralspeedy, 2024-05-17:
Well, I think I am going to have to abandon using this project. As of yesterday now Assistant is completely unusable and even if it does hear what I said it takes 5-10 seconds to eventually just say "Oops, something went wrong". No issues with a wired connection though.

I am going to order an AAWireless dongle from Amazon this weekend and see what happens because this all really seems like an Android Auto/Google App issue related to specific phones. I don't see anything in the code of this project that could possibly cause these issues.

admiralspeedy, 2024-05-23:
Been using my AAWireless dongle now for a couple days and it works flawlessly without any of the advanced settings/fixes enabled in their companion app.

So, I guess that means something about this project is causing issues between the car and the phone in regards to microphone data.

arnaullv, 2024-08-03:
Same here, with a raspberry pi zero, and 0.4 version , google assistant doest work at all. It looks it detect my voice because the horizontal color bar moves, but no order is detected (no transcrption is shown). With Android auto through cable it works without issue.

arnaullv, 2024-08-03:
I guess it could be related to bluetooth, I read that Android Auto uses bluetooth mic to comunicate

banzonam, 2025-01-17:
I've also noticed AA doesn't recognize my voice commands or is inaccurate using pi zero w and v0.4.1. With the same phone in the same car #1, back to back, I used my original AAWireless dongle with very fast and accurate voice command response. I took the pi zero w to another car #2 with the same phone, same poor voice command response. Car #2 with the original AAWireless dongle works very fast and accurate as well.

Config:
Samsung S23+, v S916USQS5CXJX
AA v13.3.644654

Raspberry Pi Zero W
WirelessAADongle v0.4.1

Car#1, Boss BE7ACP + AAWireless original hardware
Car#2, Ford Focus ST + AAWireless original hardware

manio, 2025-01-17:
@banzonam
you're using Pi Zero W, not a Pi Zero 2 W?

banzonam, 2025-01-17:
> [@banzonam](https://github.com/banzonam) you're using Pi Zero W, not a Pi Zero 2 W?

Correct, the older pi zero w.