nisargjhaveri/WirelessAndroidAutoDongle #86: Latency and choppy performance

BillBallong, 2024-03-18:
Hey! 

Setup:
Raspberry pi zero 2 w
Oneplus 9 pro
VW caddy 2019

I am running into some issues where after a while of being connected to the android auto in my car. The latency on my sound gets increasingly worse and worse starting at 1s then moving up all the way to 5ish seconds and the google maps is very laggy. The sound cuts out from time to time "indicating" thst the latency is increasing.

Sound includes both spotify app and from other apps such as youtube etc. 

There are no issues like this when I am connecting with cable directly from phone to car. 

I usually put my rpi in the glove box and if I move it closer to my phone it seems to be not lagging as much, but still within 1m of my phone at all times.

nisargjhaveri, 2024-03-20:
There could be Wifi interference issues from what you described. You can try changing the Wifi channels or try using 5GHz wifi on RPi 4 if you have one to experiment.

BillBallong, 2024-03-20:
I was doing a bit of testing today actually, and keeping it within 10 cm of the phone does seem to help a bit, it takes slightly longer for the sound to get delayed but it still occurs. I also noticed my phone saying "low quality" in the wi-fi tab in settings.

I do not have an RPi 4 to test with, but how would I go about changing the wifi channels? I was thinking about changing the country code, but I forgot until you mentioned the wifi channels. Could this affect the performance?

nisargjhaveri, 2024-03-23:
You need to update the /etc/hostapd.conf file for both country and channel.

BillBallong, 2024-03-23:
I just did today. I haven't tested it yet but I will have the chance to do that from next monday. I changed the country code to the right country and the channel to 11.

Suhailakl, 2024-06-23:
@BillBallong Was your problem solved by changing channel to 11? I'm also facing the same issue, but not as frequently.

BillBallong, 2024-06-23:
> @BillBallong Was your problem solved by changing channel to 11? I'm also facing the same issue, but not as frequently.

It did help, especially in populated areas it seemed. Also using a proper usb-data cable and not a cheap basic one I saw it being more stable. 

I also did change the country code to my respective country. 

After one of the recent updates it started to cut the audio every second or so, so I stopped using it a while ago. I think it has something to do with my sd-cards, but I haven't looked into it because lack of time. 

@Suhailakl

Suhailakl, 2024-06-27:
@BillBallong  Do we need to connect the power port of the Raspberry Pi as well? Currently, I'm only connecting to the data port. Will it affect the performance? I'm experiencing intermittent sound issues.

BillBallong, 2024-06-27:
> @BillBallong  Do we need to connect the power port of the Raspberry Pi as well? Currently, I'm only connecting to the data port. Will it affect the performance? I'm experiencing intermittent sound issues. 

I haven't been using the power port and it was working for a while without problems, it should be able to feed enough power from the data port.

Suhailakl, 2024-06-27:
@nisargjhaveri  What are your thoughts about this? Do you recommend using the power port too, or is the data port enough?

nisargjhaveri, 2024-07-02:
> @nisargjhaveri What are your thoughts about this? Do you recommend using the power port too, or is the data port enough?

Only the OTG port should generally be enough. But some people have been reporting that adding power port helps. Maybe if the data port on the car is not able to provide enough power? Not sure. I've been using both RPi Zero 2W and RPi 4 just on the OTG cable without any issues.

lob-star, 2024-07-12:
Can confirm my RPI Zero 2 W is overheating and boot looping. Tested two different devices. Symptoms start out as stuttering audio and slow interface response. About 10 mins later I get a reboot and then they continue with increasing frequency until about 10 mins later I need to unplug it to let it cool down. The copper heatsink is very hot to the touch. I also have a heatsink case which gives me a few more mins before the issues start again. I switch back to the Motorola M1A adapater and have no issues.